moonlight-stream/moonlight-embedded #255: Building moonlight for cubox-i on ArchLinux

stinuz, 2015-11-28:
Im a trying to build moonlight-embedded on archLinux for my cubox-i4 Pro. I have installed all dependencies mentioned in the wiki but i get the following error regarding broadcom & freescale: 

[alarm@alarm build]$ cmake ../
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY)
-- Could NOT find Freescale (missing:  KERNEL_INCLUDE_DIR)
-- Checking for module 'sdl2>=2.0.4'
--   Package 'sdl2>=2.0.4' not found
CMake Error at CMakeLists.txt:43 (message):
  No video output available

What depedencies am i missing and how can i get them ?

Thanks for your help

irtimmer, 2015-11-28:
It looks like you're missing the linux-headers-imx6-cubox-dt package

closed 2015-11-28 by irtimmer

stinuz, 2015-11-28:
That's it. Thanks for your very fast answer and have a nice weekend!