moonlight-stream/moonlight-embedded #256: Alsa error code -2

Gehilfe, 2015-11-29:
Hello everybody,

I´ve an problem starting moonlight-embedded on my RPi since the last Update of OSMC yesterday evening (see attached picture).
[image]

I have researched already in some forums but found nothing suitable.

The command `<aplay -l>` returns:
`<aplay: device_list:268: no soundcards found...`>

and `<lsmod`> shows me:
```
Module                  Size  Used by
8021q                  22147  0
garp                    7622  1 8021q
stp                     2430  1 garp
llc                     6031  2 stp,garp
bnep                   12310  2
bcm2835_thermal         2844  0
bcm2835_gpiomem         4026  0
bcm2835_wdt             4081  0
btusb                  28934  0
btrtl                   4596  1 btusb
btbcm                   7686  1 btusb
btintel                 3772  1 btusb
bluetooth             418101  24 bnep,btbcm,btrtl,btusb,btintel
rfkill                 22502  3 bluetooth
uio_pdrv_genirq         3885  0
uio                    10364  1 uio_pdrv_genirq
snd_bcm2835            24618  0
snd_pcm               115582  1 snd_bcm2835
snd_timer              24969  1 snd_pcm
snd                    77898  3 snd_bcm2835,snd_timer,snd_pcm
fuse                  106878  1
ipv6                  441571  42
```

The file `</etc/modules`> contains the entry
`<snd-bcm2835`>
but moonlight doesn´t start.

In OSMC the sound over HDMI still works fine.

Can anybody help me further?
Thanks and greets
Gehilfe

Gehilfe, 2015-11-29:
After rebooting I saw another error message:
[image]

Gehilfe, 2015-11-29:
Problem solved!
With command
`<COLUMNS=200 dpkg-query -l > packages_list.list`>
I created a list of all installed packages. 4 of the packages had status "rc" (reinstall needed).
One of them was `<libflac8`>. After reinstalling them everything works fine again.

closed 2015-11-29 by Gehilfe

diegombeltran, 2015-12-01:
Still not working, I'm getting the same error. Geforce Experience 2.8 beta here

Gehilfe, 2015-12-01:
Can you post your packages_list.list content?

diegombeltran, 2015-12-01:
Sure!

What I needed to do was install (not reinstall) those packages marked as RC. Unfortunately, libsdl2 has no package candidate, which I don't think is a problem as I've installed libsdl2-2.0
[paquetes.txt](https://github.com/irtimmer/moonlight-embedded/files/48771/paquetes.txt)

Gehilfe, 2015-12-01:
Try to install rbp2-image-4.2.3-2-osmc instead rbp2-image-4.3.0-10-osmc

diegombeltran, 2015-12-01:
It works! Why does it happen? I thought I was using kernel 4.3

Gehilfe, 2015-12-01:
It seems that there´s a problem with ALSA in kernel 4.3.x. I have not brought it to work with this kernel. The main thing it works again ;)

goofysmurf, 2015-12-02:
Hello Gehilfe

I noticed you are talking about the kernel image.  Just to confirm will the October osmc release have kernel 4.2.3 still or must I look at the earlier builds.

Gehilfe, 2015-12-02:
I´m not sure but I think the october osmc has kernel 4.2.3. But if you´ve already installed the november release or updated to this, you can simple install the old kernel with 
`sudo apt-get install  rbp2-image-4.2.3-2-osmc`
But it is important that you deactivate auto update in osmc settings.

diegombeltran, 2015-12-02:
Anyway, both kernels can coexist, can't they? Just do boot 4.2.3 until it works with Alsa. If I remember correctly, I didn't delete 4.3 while installing 4.2.3.

Gehilfe, 2015-12-02:
Yes, both can coexist. 
Did you know how to choose explicit one kernel to boot in osmc (e.g. like in grub)? I can find no comparable option in osmc

diegombeltran, 2015-12-03:
I think the RPi boots from the /boot partition. There are several files in that directory. Renaming the desired kernel image to **kernel.img** and putting in there **should** be enough. Though, It seems the kernel is overwritten each installation, so they do not coexist.

Gehilfe, 2015-12-03:
In order to avoid an unwanted update you can protect the kernel with 
`sudo apt-mark hold rbp2-kernel-osmc`