moonlight-stream/moonlight-embedded #264: Wired ps3 controller on raspberry pi has 2 (sticky and reversed) Dpad buttons

devidation, 2015-12-07:
Hello,

I have tried getting my ps3 controller to work following this guide: http://embedded-software-architecture.com/?page_id=147

Although I skipped the wireless stuff, I got it to work but only partially. I have tried mapping every button and when i launch steam for example, all buttons work. The only problem is that the up and left button on my Dpad work but the down and right button are also up and down and they stick. this means that when I press left, I will go left, but when I press right, I'll go left and it keeps going left even when i release the button, only stopping when pressing left.

I don't know if i'm missing something that has to be done if you wan't usb only or if the installation messed up. Any help would be appreciated.

Thanks in advance!

yakka34, 2015-12-27:
I'm having the same or similar problem with ps3 wired. left and up works but once I press right it keeps on going right indefinitely. Only way to go left is to hold down right button but after releasing it goes back to the right. I have xbox one controller connected also and it works fine until one hits the ps3 right button.

larfingshnew, 2016-02-01:
I'm having the same issue as yakka34
everything seems to work fine except for d'pad right or down.
if I press right or down it keeps looping and i can only stop it by pressing right or down a couple times again. 

I used these instructions
http://www.raspians.com/Knowledgebase/ps3-dualshock-controller-install-on-the-raspberry-pi/
however mapping to /dev/input/js0 doesn't work for me
i can only map to /dev/input/event2

duduke, 2016-03-03:
Same here, a Nyko Raven PS3 controller (not original).
When pressing right on the d-pad, it keeps sticking to the right and doesn't let go.

wouterverduin, 2016-03-25:
Any resolution for this yet?

duduke, 2016-03-25:
I was able to solve it by combining the PS3.map that comes with moonlight together with one that I've generated using the command line.

wouterverduin, 2016-03-25:
Where can i find the one provided by moonlight? 

Did you use -input to select your controller to prevent other inputs from interferring?

duduke, 2016-03-25:
https://github.com/irtimmer/moonlight-embedded/blob/master/mappings/dualshock3.conf

I didn't use the input parameter.

wouterverduin, 2016-03-25:
Thanks for the reply. Figured it out eventually. For the people who are still struggling:

**To find my USB device:**
cat /proc/bus/input/devices

**How i start a stream**:
moonlight stream -720 -30fps -app "Steam" -mapping /home/pi/play3.map -input /dev/input/event0

**My config** (what seemed to matter are the three in bold, they needed to be 0):
abs_x = 0
abs_y = 1
abs_z = 50
reverse_x = false
reverse_y = true
abs_rx = 2
abs_ry = 5
abs_rz = 51
reverse_rx = false
reverse_ry = true
**abs_deadzone = 0
abs_dpad_x = 0
abs_dpad_y = 0**
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 300
btn_east = 303
btn_south = 301
btn_west = 302
btn_select = 288
btn_start = 291
btn_mode = 704
btn_thumbl = 289
btn_thumbr = 290
btn_tl = 296
btn_tr = 297
btn_tl2 = 298
btn_tr2 = 299
btn_dpad_up = 292
btn_dpad_down = 294
btn_dpad_left = 295
btn_dpad_right = 293

Have fun people :)

closed 2016-04-03 by irtimmer