moonlight-stream/moonlight-embedded #266: Enhancement : Make so that moonlight can handle both a button index or a button code

substring, 2015-12-08:
Several (if not most) SDL2 projects usually handle a button/axis/hat index instead of a code. It makes moonlight integration complicated when you want to transfer some joypads configuration from a system to moonlight without having to map through moonlight.

Ex : https://github.com/gabomdq/SDL_GameControllerDB -> uses the indexes istead of codes (see https://wiki.libsdl.org/SDL_GameControllerAddMapping)
projects involving retroarch work the same

substring, 2016-01-25:
@rockaddicted @digitallumberjack
was solved with 3.3.0beta16 or 17. You can close it

irtimmer, 2017-05-27:
It took some time but since commit 9f43712fc89ab6e56a5a6f777deeb5a7c5fc7192 a SDL like mapping is used.

closed 2017-05-27 by irtimmer