moonlight-stream/moonlight-embedded #267: mDNS discovery

substring, 2015-12-09:
Hi,

on the main page, it is said "Use mDNS to scan for compatible GeForce Experience (GFE) machines on the network"

How does it work from command line ? Running `moonlight pair` just freezes, and I see no other option to discover GFE servers

Haris1977, 2015-12-09:
Does

moonlight pair IP_OF_PC work?

substring, 2015-12-09:
Yes it does ! I expected something more like android moonlight : an option that lists available GFE servers. At least that's how i understand the text i quoted

Haris1977, 2015-12-09:
Good to hear it works. Just a question though: Do you have any fps drops with your games? What is your host pc? I have opened an issue here https://github.com/irtimmer/moonlight-embedded/issues/252 and i am gathering some thoughts..

substring, 2015-12-09:
i'll answer on your issue not to pollute this one

irtimmer, 2016-04-03:
Without specifying the ip adres Moonlight Embedded use the first discovered computer. Too ensure discovering is working the Avahi daemon needs to be running.

closed 2016-04-03 by irtimmer