moonlight-stream/moonlight-embedded #269: Two controllers input (not) working?

Haris1977, 2015-12-09:
By using the command:

moonlight stream -mapping xbox.map -input /dev/input/event2 -mapping logitech.map -input /dev/input/event3 -input /dev/input/event0 -input /dev/input/event1 192.168.1.109  (events0 and 1 are my keyboard/mouse wireless combo)

inside steam i can use both controllers, but in game (e.g Nba2k10) only one controller and keyboard are present. Is this a known bug?

3djake, 2016-05-12:
Yeah, I am having issues with it too, how ever I cannot even use steam big picture mode for some reason. 
It used to work with one controller, I just got a second for some multiplayer games but I have not used moonlight in a few months and cannot remember how I got the controller working from looking at history.

I use 2 Wireless Logitech F710's and xbian. My issue is more likely PEBCAK but any advice on getting them working would be appreciated.

`sudo xboxdrv -i 0 --silent & sudo xboxdrv -i 1 --silent & moonlight stream -mapping testx -input /dev/input/event0 -mapping testx -input /dev/input/event1 -app Steam 10.0.0.30`

RikJm, 2016-05-30:
Check if after you've clicked the controller icon in big picture (from your computer using a mouse) the controllers work again, I've been having this issue lately and "waking it up" from my pc seems to work.

Like you I had them working great on an earlier build but now it doesn't seem to pick them to be the default input for some reason.

closed 2017-03-18 by irtimmer