moonlight-stream/moonlight-embedded #270: No 5.1, low audio level

PhinPhin, 2015-12-10:
Hi, I use Moonlight on an RPi with the latest Raspbian release.

So far I got Battlefront running by adding it to the Steam-library (I can't seem, to find a way to type that trademark-char that is in the gamename to start it directly). Though With or without the -surround flag I don't get any 5.1 sound. My AV-receiver only says 2.0 PCM. Do I have to manually set anything? On my PC i mostly use headphones. Do I need to set something? 

Apart from that, the audio-volume is ridicules low, I have to turn up the AVR to 80% to actually hear anything, at that volume normally you would get bloody ears, that's not ok... if somebody switches audiosources it could kill the speakers. Please tell me a fix for this!

pengc99, 2015-12-13:
you may need to set the volume of the output to 0dB (100%) - in the command line enter this:

amixer set PCM 100

This fixed the volume issue for me.

PhinPhin, 2015-12-13:
Ah thanks, I will try this. Will thise set the volume once for will it saved even over a reboot?

I still wonder why I don't get any Surround, though. Is there any documentation what gets transmitted if I use the -surround flag?

pengc99, 2015-12-14:
I put that setting in my rc.local so it gets set every time on boot along with the initialization of my Xbox wireless controller driver.

KillerXtreme-, 2015-12-20:
I am having issue with 5.1 Surround as well. I'm launching moonlight through RetroPi/EmulationStation and it seems to be stuck on 2ch Stereo. I've tried compiling the latest source as well as having the latest GFE and I'm unable to get surround going with Borderlands 2, which is one of the supported 5.1 surround games as stated by nvidia.

PhinPhin, 2015-12-22:
Haven't found a solution yet.... :/

Xiller, 2015-12-31:
Same here with FireTV (old good one :> ) only 2ch Stereo. [movies over amazone prime are 5.1]

PhinPhin, 2016-01-01:
Had an odd issue yesterday: everytime I used the surroundflag the game crashed as soon the gamewindow appeared above the Gamestream-window(before it normally would go fullscreen). :(

irtimmer, 2016-02-19:
Could you check if version 2.1.3 solves the 5.1 surround problem.

irtimmer, 2016-03-25:
The Raspberry Pi couldn't output 5.1 audio using the ALSA backend (unless you used a usb audiocard). With the addition of the OMX audio backend this should be solved by commit 8de4fe1cf31ef32ab7cc1ece00042e92b5c85d15 which will be included in the next release

closed 2016-03-25 by irtimmer

PhinPhin, 2016-06-15:
Haven't checked yet, sorry. Though I really wonder why the HDMI-implemenation is so limited...

KillerXtreme-, 2016-06-24:
@irtimmer is there a way to do audio passthrough for surround sound through HDMI? I'm able to utilize my surround sound system while watching movies through the pi, I'm sure there has to be a way to use passthrough for moonlight as well.

irtimmer, 2016-06-24:
You can add the parameter "-audio hdmi" to your moonlight command

PhinPhin, 2016-08-05:
Need to check that, atleast it's 3 months later after all :P