moonlight-stream/moonlight-embedded #272: Cant connect to server x.x.x.x

Kheiryk, 2015-12-13:
Hello, 
I installed the last packet Moonlight on my rpi2 (Raspbian Jessie) and i've GFE beta 2.8.0.80.
When i want to pair Moonlight with moonlight pair x.x.x.x, i've the message :
Cant connect to server x.x.x.x
My firewall on my pc, not inform me to a connexion is trying to open.
Maybe a version GFE problem ? any suggestions ?
thanks

Haris1977, 2015-12-13:
Please install gfe 2.7.4.10 to see if problems persists. I think it is nvidia's fault.

Kheiryk, 2015-12-13:
Yes, it's works, thanks a lot !
I continue my configuration

Kheiryk, 2015-12-13:
another question, i can start a game, and the xbox controller works apparently but when i quit a game, i've the display "Gamestream", the game is close, but i must go to my computer and stop this screen (alt+f4).

Haris1977, 2015-12-13:
Do you start steam or individual app? This has happened to me when quiting individual app and not steam big picture..The problem is when you exit the game, it hangs on the Gamestream image..

Kheiryk, 2015-12-13:
individual app yes

Haris1977, 2015-12-13:
I have the same problem, i think it is moonlight embedded bug

Kheiryk, 2015-12-13:
ok thank you :)

Haris1977, 2015-12-13:
You are welcome

@Kheiryk do you face any fps drop when you stream games? Have opened an issue here #272 and i am collecting some opinions in order to eliminate some errors. 

Thanks

closed 2016-04-03 by irtimmer