moonlight-stream/moonlight-embedded #274: Stuck at Starting RTSP handshake

devidation, 2015-12-13:
Hello,
I have tried streaming for some days now and this issue is quite annoying.
When i start the stream, it goes to Starting RTSP handshake... and just keeps trying to handshake.
I have however been able to succesfully stream several times. Sometimes it works and sometimes it won't whatever I try. Is there something I'm missing that's preventing this from going through from time to time? Also do you need Steam and Geforce Experience to be open as in the window has to be visible?
Thanks in advance.

H3llsing, 2015-12-13:
try reinstalling Geforce Experience and nvidia drivers:
same issue as here: https://github.com/irtimmer/moonlight-embedded/issues/212

devidation, 2015-12-13:
Thanks H3llsing, I have done that and everything seems to work now. Only thing that I have trouble with is something I'm not sure of. It could be 2 things. The games that are not launching initially are either non steam games or games which have the keyboard required icon I'm not sure why. Because after several attempts I quit steam and then suddenly one of these problematic games launched lol.
But Atleast I can stream now so thanks again.

H3llsing, 2015-12-21:
Have you tried to connect an USB keyboard to your pi? Are these games lunching then?
What is happening if you try to lunch them? Is there any error in the console?

closed 2016-01-16 by irtimmer