moonlight-stream/moonlight-embedded #275: gamestream window crashes when starting steam, sticks on rtsp handshake

tooandrew, 2015-12-14:
when attempting to stream steam in big picture mode from my windows 10 desktop, the gamestream window opens, but crashes immediately, and the pi attempts to start the rtsp handshake for a long time then fails with error 110:

Initializing platform...done
Resolving host name...done
Starting RTSP handshake...connect() failed: 110
RTSP OPTIONS request failed: 110
failed: 110
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16

however, when i stream assassin's creed syndicate ( moonlight stream -app ACS ), the stream is initialized just fine, the gamestream window stays open and everything works.

H3llsing, 2015-12-22:
use the search first!
Have you tried this ---> 
https://github.com/irtimmer/moonlight-embedded/issues/274  ?

Athanaetos, 2015-12-24:
Tried what?

irtimmer, 2016-01-01:
Duplicate of #274

closed 2016-01-01 by irtimmer