moonlight-stream/moonlight-embedded #276: A couple of issues and possible thoughts for a solution.

KillerXtreme-, 2015-12-20:
Firstly I don't know why this is happening, I don't know if its the new version of GFE or moonlight. However when I start up Steam with -1080 on my television it no longer fills the full screen. Its almost as if it leaves bars on the left and right side of the screen. (in which you can see terminal output on the left side) As well with Steam Big Picture it ends up cropping the top as well. No Idea whats going on there.

I'm also trying to get 5.1 Surround to work. However it seems that moonlight is always using ALSA, in which alsa seems to only work with stereo? So its taking the 5.1 Surround and resampling it to a stereo output. This slightly confuses me as I'm able to output 7.1 surround no problem using Kodi while watching movies. So why is it that it won't output 5.1. My receiver more than happly accepts most if not all signals. I'm wondering if its because the pi itself is decoding the audio signal. So that makes me wonder if you can implement a passthrough so a receiver can decode it through HDMI.

If anyone can assist with these issues that'd be amazing!

HarryLovesCode, 2015-12-28:
@KillerXtreme-

1. Does it properly fill up the screen when you use `-720`? Have you disabled overscan and changed TV settings to automatically adjust to the image size?
2. I can confirm the 5.1 surround sound issue. I ended up sticking with stereo because I was having weird issues with games audio being very soft / not functioning at all. There are a few issues filed for it on this repo.

KillerXtreme-, 2016-01-05:
To answer your #1 No it doesn't. Like I said I get a terminal behind the nvidia stream that takes up the full screen. However the nvidia gamestream seems to be 4:3 even with a -720 or -1080 flag. 

For your #2 there is a command to set alsa (the Pi's audio BS) to 100% audio volume. You can use alsamixer command to change the volume, but there was a command that changed it to 100 that worked for me.

irtimmer, 2016-04-03:
5.1 surround should work with newer versions of Moonlight Embedded and the 4:3 is propably is a Gamestream bug and is hopefully fixed in a newer release.

closed 2016-04-03 by irtimmer

Sardtok, 2016-04-03:
By the way, I managed to get a game to play in fullscreen. I had moonlight running with the default resolution (i.e. -720), and switched the resolution in-game to 1080p. On other resolutions, it would not fill the screen. In 1920x1200 (my server monitor's native resolution), I got black bars all around. In 720p I got black bars on the sides (so the image was slightly squashed). I always empty the console before I start playing, but if I don't the Pi boot sequence output shows on the left side. So it's worth experimenting with resolutions on both host and Pi.