moonlight-stream/moonlight-embedded #277: Streaming Server Reboots without Warning

mdjones, 2015-12-22:
Hi, 

Not sure if this is a moonlight-embedded issue or something strange about my PC. I get a stream running just fine and then after a random amount of time the screen freezes on the Pi. If I go back to the streaming server I see that is has gone through a power cycle and is waiting for me to log back in. I have so far only seen this with Fallout 4. 

I thought this might have something to do with the screen turning off (Maybe moonlight doesn't like headless) from the power settings. However, my screen is set to turn off at 10 minutes and I have been able to play for much longer then 10 minutes. 

Any ideas out there?

----------------------------------------------------------------------------------------------

CMD: 
/home/pi/game_stream/moonlight-embedded-2.1.2/build/moonlight stream 192.168.1.141 -mapping /home/pi/Moonlight/xbox.map -720 -30fps -app Steam

PI: Linux raspberrypi 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l

Server: Windows 10, 64 Bit. NVIDIA 960 (GFORCE GTX)

mdjones, 2015-12-24:
Update: If I leave the monitor on and set the display on the server to never shut off it runs for much longer. Random disconnects or moonlight crashes still occur but server does not reboot.

7hunderbug, 2016-01-04:
Sounds like a problem with your PC and I suspect your Nvidia driver and GFE install. Reinstall both using a clean install from the custom option in the installer.

Unknownforce351, 2016-01-18:
I would install some monitoring tools and make sure you're not hitting some bad temperatures on the PC.   I've never had my PC restart with moonlight.  I did have lots of crashing issues with moonlight when I first started with moonlight on my Raspberry Pi, but it seemed to be resolved when I updated moonlight to the latest git pull and put in a hard wire ethernet rather than using WiFi.  I never crash anymore, I just get some artifacting here and there, usually after streaming for a longer period of time.  I also play at 1080p and 60fps, without any issues, very slight audio/input lag, but only noticeable by the very trained eye.

Like I said, check your PC, make sure it's got proper cooling and that real temperatures reflect that.  For a very crude test, pull the side of the case off and try it again, see how long it lasts.  You can suppliment that test by adding a small fan aimed at the inside of the case.  If it lasts longer and/or indefinitely, then you know it's a temperature issue.

I also suggest using a hard wire ethernet cable instead of WiFi.  It makes a WORLD of difference.

Good luck!

mdjones, 2016-01-22:
Thanks @Unknownforce351 I always use Ethernet. I will check the temperature. Do you have a recommendation for monitoring tools?

@7hunderbug I did do a clean install back then but for me just leaving the monitor on seemed to fix this problem.

Unknownforce351, 2016-01-27:
I personally use Open Hardware Monitor.  http://openhardwaremonitor.org/

It's free, open source, portable and easy to use.  But there are plenty of other monitoring tools out there, if it doesn't appeal to you.

closed 2016-04-03 by irtimmer