moonlight-stream/moonlight-embedded #278: Run in PiTFT but error

sabas1080, 2015-12-24:
Hi

I have Raspberry Pi with Jessie and  PiTFT 2.8" TFT 320x240 + Capacitive Touchscreen - Raspberry Pi Model B of Adafruit (https://www.adafruit.com/products/1983)

I want run moonlight in PiTFT, but i have next error

alsa error from write: -32
alsa error from write: -32
alsa error from write: -32
Alsa Shortm write, write 6 frames

Thanks

Sardtok, 2015-12-31:
I believe that's a server side issue. At least it was when it happened to me. Try restarting GeForce Experience, or rebooting the streaming computer.

Ranger63, 2016-01-02:
I have a different screen but also get that error. I was running into that problem when my Pi stopped detecting my audio output devices, which I fixed by just running a bash whenever I boot up. 
Also, my screens probably different, but when I stream to my pi with the Pi touchscreen, the video outputs to HDMI.

ihavenick, 2016-01-08:
I am trying with another screen , but i only seeing terminal .
Pressing buttons on gamepad works on server but there is no video on client. 
I using spi lcd touch screen . And touch screen works too but no video.

sabas1080, 2016-01-08:
@ihavenick the same for me

Ranger63, 2016-01-08:
I've found that moonlight can only output to hdmi and maybe composit ( haven't tested composit), and after talking to some of the community it looks like that's because of hardware video acceleration. I found that when I was using the touchscreen LCD on pi, moonlight would output to hdmi anyways.

irtimmer, 2016-04-03:
Video hardware acceleration is only available trough HDMI and composite on the Raspberry Pi.

closed 2016-04-03 by irtimmer