moonlight-stream/moonlight-embedded #280: Audio lag

Tuxliri, 2015-12-29:
On the raspberry pi 2 the audio lags a little, it's reproduced about half a second after the action. Any idea on how to solve this?

fadetonoir, 2016-01-01:
I am having the same issue. How much RAM do you have allocated to your GPU? After a clean installation NOOBS 1.5.0 on my Raspberry Pi 2 and then installing nothing else but Moonlight I had a noticeable audio lag of about half a second, the same as you. I have since changed my GPU RAM allocation to 256mb and this seems to have reduced, but not completely eliminated, the audio lag. (I was previously trying to run Moonlight on my other Raspberry Pi, a Model B, where the audio stutters badly, making it inaudible, and produces a constant stream of OOS errors in the Pi's terminal. I couldn't improve on that in any way, either by changing the amount of GPU RAM or overclocking the Pi.)

Tuxliri, 2016-01-03:
I've tried the same thing with similar effects, I'm using the latest recalbox release. I'll try tomorrow to test more, bow can I log what's happening?

Unknownforce351, 2016-01-07:
I had this problem as well.  I actually found a setting on my TV, strangely enough, that fixed this in my case... My TV is an LG 55" TV, and it has a setting in the sound options called AV Sync Adjust.  It has an adjustable setting to actually go from -5 to +15, which, I'm not sure exactly what scale these numbers represent, but it definitely moves the audio playback synchronization ahead or behind by a certain degree...  Well the audio is AHEAD of the video (too slow/late), so I set to -5, to bring it back in line, and viola, audio lag is gone!

Just figured I'd share my experience with it.  It's probably still a real issue as far as programming/development side is concerned, but I'm glad it works with this setting on the TV anyway.  You might look for something similar on your TV, if possible.

Tuxliri, 2016-01-16:
I don't have that option in my TV, I can't try

squidgydude, 2016-04-04:
I've found the lag much improved, if not resolved, in the very latest version of moonlight (v2.2) and Geforce Experience.

lexa2, 2016-04-26:
It seems this issue is a "floating one". At least for me. I've got RPI2 with OSMC installed, GPU mem allocation is 256Mb. To start up moonlight I'm using alpha version of Kodi plugin Luna. Streaming settings are 1080p/60FPS/Auto net bandwidth. Sometimes I do get audio lag, sometimes not. I hadn't been able to figure out any consistent pattern for lag to appear. I.e. it could be something like I start up moonlight and play for some time without issues. Then I shut down streaming using Ctrl+Alt+Shift+Q, use Kodi on RPi2 to watch some movies and then attach back to the game streaming by launching moonlight with the same app profile and suddenly audio lag appears. If I detach and re-attach from streaming again - it is equally possible for audio lag to be there or to be lag-free experience.

I'm not sure that the issue I have is the same as this one but IMO it is worth to mention that there are people out there who at least sometimes get no audio lag with moonlight and RPi2 - i.e. it is at least possible with this hardware and current moonlight codebase.

benwilesmusic, 2017-09-29:
As posted on #517:

I'm not sure this is a duplicate; #280 seems to refer to an audio lag that sometimes happens when starting moonlight, while this one refers to a sync problem whereby the audio drifts: while the audio is in sync at the start of the stream, it gets further out of sync as the stream continues.

I have observed this latter problem.

jarredgallina, 2019-07-26:
This is also happening on the Raspberry Pi 4. Audio is in sync when the stream starts, and gets out of sync over time. It becomes noticeable after around 20m, and is pretty much unbearable after a couple of hours.

zartinn, 2020-03-17:
Have the same experience as jarredgallina on my Pi 4.
Log does not show any errors/problems :/

exodious, 2021-01-06:
Having the same issue as the previous two posts. Been playing The Outer Worlds. It’s most noticeable in the in-game menus like inventory etc. There’s a sound that gets played when switching screens, and after a few minutes of streaming the delay between switching and the sound playing becomes very noticeable. Also noticeable when talking to NPCs or firing ranged weapons.

If I leave the game running and exit moonlight, then restart it, the audio is synced back up for a short time.

nicolaspernoud, 2021-02-02:
I have the same problem with raspberry pi 4 : audio starts fine and lag slowly increases, until not playable.

exodious, 2021-02-03:
Moonlight-qt for Raspberry Pi 4 is still in beta, but I don't have the audio lag as with moonlight-embedded. https://github.com/moonlight-stream/moonlight-docs/wiki/Installing-Moonlight-Qt-on-Raspberry-Pi-4

nicolaspernoud, 2021-02-04:
@exodious : I prefer to use embedded since I need a command line utility... But thanks anyway.

rivasd, 2021-04-11:
Can confirm I have the same slow audio drift on raspberry pi 4 with Retropie

MikeAtRoot, 2021-09-14:
All,

In 2020, after numerous testing and debugging attempts, I've determined certain parameters exclusively set or commented out within moonlight.conf will resolve this annoying audio issue. Please note I'm using moonlight-embedded via wifi with a Raspberry Pi 4.

Mainly, I believe explicitly setting the 'audio' parameter to 'sysdefault' is what resolves this bug. However, I'll annotate the moonlight.conf configuration that I personally use at the bottom of this post. Validating that this issue is solved with the configuration file I provide would be appreciated. Also, I had to comment out the platform parameter, as I think even setting 'platform' manually to 'auto' or 'pi' would still cause an increasing audio delay over time.

It is disappointing that the developers of this project did not address this bug that has been documented for nearly six years now, nor is there any clarity or concrete documentation as to why explicitly setting default parameters as opposed to leaving them commented out would even make a difference. However, I hope the config file I detail below will resolve this issue, and we can finally put this to bed.

Thanks,
-Mike

-----
## Hostname or IP-address of host to connect to
## By default host is autodiscovered using mDNS
#address = 1.2.3.4

## Video streaming configuration
width = 1920
height = 1080
fps = 60

## Bitrate depends by default on resolution and fps
## Set to -1 to enable default
## 20Mbps (20000) for 1080p (60 fps)
## 10Mbps (10000) for 1080p or 60 fps
## 5Mbps (5000) for lower resolution or fps
bitrate = -1

## Size of network packets should be lower than MTU
packetsize = 1024

## Select video codec (auto/h264/h265)
codec = auto

## Default started application on host
#app = Steam

## Default used mapping for streaming
## Searched for in $XDG_DATA_DIRS/moonlight or /usr/share/moonlight and /usr/local/share/moonlight
## Mapping can also be user overrided in $XDG_CONFIG_DIR/moonlight or ~/.config/moonlight or current directory
mapping = gamecontrollerdb.txt

## Enable selected input devices
## By default all available input devices should be used
## Only evdev devices /dev/input/event* are allowed
## To use a different mapping then default another mapping should be declared above the input
input = /dev/input/event0

## Enable GFE for changing graphical game settings for optimal performance and quality
#sops = true

## Play audio on host instead of streaming to client
#localaudio = false

## Send quit app request to remote after quitting session
#quitappafter = false

## Select audio device to play sound on
audio = sysdefault

## Select the audio and video decoder to use
## default - autodetect
## aml - hardware video decoder for ODROID-C1/C2
## rk  - hardware video decoder for ODROID-N1 Rockchip
## omx - hardware video decoder for Raspberry Pi
## imx - hardware video decoder for i.MX6 devices
## x11 - software decoder
## sdl - software decoder with SDL input and audio
## fake - no audio and video
#platform = pi

## Directory to store encryption keys
## By default keys are stored in $XDG_CACHE_DIR/moonlight or ~/.cache/moonlight
#keydir = /dir/to/keys

## Enable QOS settings to optimize for internet instead of local network
#remote = false

## Enable 5.1 surround sound
#surround = false

## Load additional configuration files
#config = /path/to/config

marionovelles, 2024-01-19:
My device suffer from the same audio delay problem that increases over time.
I tried the fix that the last comment wrote, but in my device this change makes moonlight crash. So it's not a solution for everyone.