moonlight-stream/moonlight-embedded #282: Stream won't start OSMC - audio problem?

diagonali, 2016-01-01:
After setting up a clean OSMC installation on a Raspberry Pi 1

adding the following sources: deb http://archive.itimmer.nl/raspbian/moonlight jessie main

And installing in the usual way

It pairs successfully but when I try to stream i get this, which I've never seen before in previous setups:

osmc@osmc:/$ moonlight stream 192.168.0.90
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4259:(sndconfig_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4259:(sndconfig_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4259:(sndconfig_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM sysdefault
Alsa error code -2

Which seems to point to an audio error of some sort. Not sure how on a completely fresh install but wondering if anyone knows what's causing this?

Also, when I type:

cat /proc/asound/cards

I'm getting:

--- no soundcards ---

I've tried sudo modprobe snd-bcm2835 and it's still not working. Is OSMC set up differently since last time I tried?

7hunderbug, 2016-01-04:
The osmc user account on OSMC may not be part of the audio group or have the correct permissions to access the audio device. Try running moonlight as root or with sudo.

7hunderbug, 2016-01-08:
Also, check this (closed) bug report - it pertains to the kernel version in OSMC and might be helpful: https://github.com/irtimmer/moonlight-embedded/issues/256

irtimmer, 2016-01-16:
Duplicate of #239

closed 2016-01-16 by irtimmer