moonlight-stream/moonlight-embedded #283: Can't find app list

choupal, 2016-01-01:
Hi everyone, I'm really sorry if my problem has already been solved but I can't find an answer anywhere.
 I set up my Rpi2 with raspbian/moonlight. I followed many differents tutorials but they all have the same result.

Everything is going just well for the installation but when I try to stream my pc I get 2 nice errors :
"Can't find app list"
"Can't find app Steam"

I searched a lot in forums and I tried everything I could find ( disable firewall or making steam an exception, adding steam to the GFE game list, tried to stream when GFE and Steam are running on the pc).

Nothing worked. (I have the required configuration for moonlight and NVidia GameStream)
I also tried a lot of different tutorials using different methods from getting the full package and install it, to getting the source code an compile it. And all of those different ways to make moonlight work had the same errors.

I'm pretty sure that there's something on my PC that is making it impossible to run but i don't know what.
I hope you can help me and thanks you in advance.

ps: sorry for the very bad english writing, I hope you will not loose your head trying to understand me :smile:

7hunderbug, 2016-01-04:
Have you tried reinstalling Geforce Experience and Nvidia drivers? Both can cause problems with streaming. I recommend a 'custom' install rather than 'express' so that you can choose the option to perform a clean install.

choupal, 2016-01-04:
Hi, thanks for your answer, i did what you told me, reinstalling last GFE and Nvidia drivers with clean install and it didn't change anything. Still get those errors.

I made some other tests with Moonlight IOS on my iPhone (even if i don't want to use it) and i get a similar error saying  "Fetching app list failed".
I tried to stream with those 2 platform with GFE open ,with GFE and Steam open, with GFE and Steam closed and i always have the same errors. I will try on mac later.

There must be something on my PC that block moonlight from getting the app list. Or something on my router.
I also only have a keyboard+mouse plug into my RPi, i don't have a controller yet. Do i need to plug a controller to make moonlight work ? Or can it work only with keyboard+mouse ?

I notice something else too, in the Shield preferences most of the tutorials i followed had an option to "Stream Games On My Own Network".
I do not have that option. 
I have "Allow this pc to stream games to Shield Devices"

I'm quite lost, i tried everything to make it work and it doesn't want to work :cry: 

Here's my GFE Screens
[image]

7hunderbug, 2016-01-04:
Very strange! Perhaps there is a bug when there are a lot of games detected by GFE, however I have 30 games detected without issue.....

Nvidia changed the options in GFE for choosing which network to stream to and now only has the option "Allow this pc to stream games to Shield Devices" so you're all right with that.

Not having a controller plugged in shouldn't affect anything - moonlight works for me with just keyboard and mouse.

Under GFE preferences, you shouldn't need to add the direct file path to Steam.exe in the "SHIELD" tab (I would remove that). GFE should automatically detect Steam as long as you have the folder path to Steam in the "Games" tab. In my GFE preferences "Games" tab I have pathes to the folders containing games and Steam like so
:
G:\Steam
G:\Games
G:\Program Files (x86)\Origin Games

And running a scan for games after adding these paths allows GFE to detect everything.

From @cgutman 's post on xdadevelopers ( http://forum.xda-developers.com/showthread.php?t=2505510 ), to test GFE, try accessing the following URLs in your browser and post the output here:

http://127.0.0.1:47989/serverinfo?uniqueid=1234
https://127.0.0.1:47984/serverinfo?uniqueid=1234

Also, try disabling Window firewall and any other third-party firewall on your PC temporarily to see if that helps.

7hunderbug, 2016-01-04:
I've just had another thought. Try recreated the moonlight gamestream certificates and keys by doing the following:

1. On the Pi, delete the folder ~/.cache/moonlight (or /home/pi/.cache/moonlight)
2. On the host PC, in GFE, click on Preferences->SHIELD->Forget devices
3. Run moonlight pair again on your Pi to recreate the moonlight gamestream certificates.
4. Try running moonlight again to see if applist is detected.

shayaantx, 2016-01-08:
I looked at https://github.com/irtimmer/moonlight-embedded/blob/70970e9b276aa298a79b441d4f40234dd93f5ea7/libgamestream/client.c

Specifically the code that will throw the error "Can't get app list" (seems like you may be running an older version of moonlight, since your message "Can't find app list", fyi) when it makes a ssl connection to the gfe server.

An example url you can use to see the failure is, http://127.0.0.1:47989/applist?uniqueid=1234 (replace hostname). If you try to access this url without the necessary client certificates (in your case, the ones in /home/pi/.cache/moonlight most likely) you will get a connection refused error.

One thing I would try is either manually creating the curl statement yourself (see https://github.com/irtimmer/moonlight-embedded/blob/af383eb37102de787a1f0266e144839ac762601d/libgamestream/http.c for how its done in the code) or modify the code that makes the these http request to print out the errors and/or warnings when making the http requests.

- Using http://curl.haxx.se/libcurl/c/CURLOPT_DEBUGFUNCTION.html will easily print out the curl statement for you.

Also does anything sit in between your pi and your host machine other than the router? (i.e., proxies, other routers maybe, etc)

closed 2017-03-18 by irtimmer

ma16v3, 2018-01-15:
Still got  the same error as before, so I moved down to 3.11, now I get "can't get app list" error. An improvement, I think. Not sure if this is a Vita issue or PC issue.