moonlight-stream/moonlight-embedded #284: How is video outputted

Ranger63, 2016-01-02:
This isn't an issue per say, but a question. I have the RPi screen found here: http://www.amazon.com/gp/product/B013E0IJVE?psc=1&redirect=true&ref_=od_aui_detailpages00
  and when I use it, HDMI is not available, however the Moonlight streaming outputs to hdmi while I am using the screen I listed, this shouldn't be possible as its my understanding the pi is told there is no hdmi. (Getting over how poorly designed the screen was), I was wondering if anyone could simply tell me how video is going to the hdmi, so I can understand it and perhaps utilize it in other ways.

7hunderbug, 2016-01-04:
Going by the reviews on Amazon related to that screen, I suspect you are not going to have any luck getting it to work. The resolution of the screen will probably also be a problem - are you sure your games can operate at 480x320 resolution? I would try running your games at that resolution on your PC first to make sure they will actually run and are playable.

If you don't intend to use moonlight for games I would suggest that there are better alternatives for remote display streaming (RDP, VNC etc).

In answer to the question of how the video works. The Raspberry Pi build defaults to using the Broadcom OMX libraries for GPU hardware accelerated video playback and as far as I know, will only output via HDMI (or possibly via composite if moonlight width and height are set to 720 and 576 respectively for PAL and the boot config.txt has the respective switches for outputting to composite. The official Pi screen may also work, but again, the resolution is so low it may be impractical.).

I believe that the screen you have linked, connects to the Pi via the GPIO/SPI header and uses the framebuffer to drive the display using the fbtft driver. https://github.com/notro/fbtft
As far as I am aware, Linux won't do hardware accelerated graphics via the framebuffer.

It may be possible to force the SDL build of moonlight on the Pi which might work but don't ask me how to do that!

This wiki for the fbtft driver might be helpful:
https://github.com/notro/fbtft/wiki

And these links from the Raspberry Pi forum:
https://www.raspberrypi.org/forums/viewtopic.php?t=79120&f=93
https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=125612

7hunderbug, 2016-01-04:
To clarify the above. It is my understanding that the Raspberry Pi GPU videocore hardware accelerated H.264 decode as used by moonlight, bypasses the Linux framebuffer entirely and so won't be able to display output to a framebuffer driven display like that small LCD. The SDL build might work.
Disclaimer: Mu understanding could be entirely incorrect!

irtimmer, 2016-01-16:
Duplicate of #278

closed 2016-01-16 by irtimmer