moonlight-stream/moonlight-embedded #286: Moonlight stream Windows to Windows

Braindeadd, 2016-01-03:
Hi all, im having trouble setting up Moonlight to my windows laptop. 

I want to stream games from my PC to my laptop. i have managed to get moonlight working with my phone(stream 60fps), but i want to use my laptop now. Does anybody knows how to setup moonlight Pc to Pc? Thanks in advance. (im new here, so im not sure this is the right location to post this).
PC-Windows 10 64bit.
Laptop-Windows 10 64bit.

Braindeadd, 2016-01-03:
some more information: i have downloaded moonlight-win64.jar from https://github.com/moonlight-stream/moonlight-pc/releases. do i need to unpack this file(And where?) to work? i just dont know what the next step is.

irtimmer, 2016-01-03:
This isn't Moonlight Pc

p.s. issues for Moonlight Pc can be added on https://github.com/moonlight-stream/moonlight-pc/issues

closed 2016-01-03 by irtimmer