moonlight-stream/moonlight-embedded #288: Xbox One Wireless controller doesn't work on games, only on Steam interface

ZeFifi, 2016-01-04:
Hi all,

I just installed Moonlight on my Raspberry Pi 2 (Raspbian). I'm using a Xbox One Wireless controller and it's works very well with Steam and with some (1/3 in fact) games. I have no control/sound/stream issues with Life is strange but when I want to play to "Ori and the blind forest" or "Splinter Cell Blacklist", the pad is not detected (but it's OK if I launch the game directly on my computer).

I read something about the rxinput.dll file from the C:\Program Files (x86)\NVIDIA Corporation\NvStreamSrv\ folders. I renamed it as .dll.old : despite a warning window from nVidia GeFore Experience, the game launches and the controller works... but I have no sound !

Did someone had the same issue and know how could I fix it :) ? 

Thanks a lot for you help !

irtimmer, 2016-01-04:
Duplicate of #171

closed 2016-01-04 by irtimmer

7hunderbug, 2016-01-04:
Is your Xbox controller connected to your PC or to your Raspberry Pi?

If connected to your Raspberry Pi, which driver are you using, xpad or xboxdrv?

If connected to your PC, moonlight might not work correctly as it expects your controller to be connected to the Pi, and all control signals passed via the gamestream protocol.

ZeFifi, 2016-01-04:
Hi @irtimmer ,

Thanks for you reply. I've already read it before posting here but it's for Xbox 360 Wireless controller. I'm using the One one ;)

@7hunderbug Thanks ! My wireless receiver is connected to my computer, not to the Raspberry. I'm at work right now but I'm wondering something : would the pad work if it's wirelessly connected to the Pi ? Don't Steam expect the controller to be connected where it's hosted, meaning on the PC ?

Thanks again guys !

7hunderbug, 2016-01-04:
Moonlight and the Nvidia gamestream protocol expect game control signals to originate from the client device, not the host PC, therefore it is preferrable to have your controller connected to your Rasperry Pi. I can't remember off hand whether moonlight has an option to force game control from the host side however, the previous statement stands.

ZeFifi, 2016-01-04:
Mmh OK ! Can't wait to try this tonight ! It would be so cool ! Do you know if the xpad or xboxdrv drivers are supposed to support the Xbox One Wireless gamepad ?

7hunderbug, 2016-01-04:
http://pingus.seul.org/~grumbel/xboxdrv/

"Supported and tested devices
    Xbox1 controller, both official and third party
    Xbox360 USB controller
    Xbox360 wireless controller via the USB wireless receiver
    Xbox360 USB guitar and drum kit
    Thrustmaster Dual Power 3 Gamepad
    any joystick support by Linux via the evdev interface"

ZeFifi, 2016-01-04:
@7hunderbug Thanks a lot, if it works... :D ! I keep you posted !

Kalagon, 2016-01-04:
If you want to use controllers on your host PC you need to rename/remove the rximput.dll in both the Program Files and Program Files (x86) nvstrmsrv folders. After only removing one of them I also didn't get any audio on the Moonlight client.
You'll need to do this every time GeForce Experience is updated, but it shouldn't be too difficult to do this with a script.
The Problem is that many games like Rocket League etc. only accept one Controller in the menus, thus choosing the first one that pops up. The Nvidia server registers the remote controller from the client as a local controller. If this one pops up first you won't be able to control it with the controllers that are actually connected locally.
Setting up the controller on the Pi should work but can be tiresome if it doesn't work as expected.

ZeFifi, 2016-01-04:
Thanks for your reply @Kalagon ! I didn't know this file was in 2 folders ! I'll check this out tonight and will keep you posted !

ZeFifi, 2016-01-05:
@Kalagon @7hunderbug  Thanks a lot for the tips, it worked ! :)

Direcka, 2016-10-19:
Renaming both DLL files is not working for me. Have someone same problem or any workaround?

Xiller, 2016-10-29:
Hi Direcka, same here, i think it is a new bug since GFE 3.1. 

i tryed to rename the files here too: 
C:\ProgramData\NVIDIA Corporation\GeForce Experience\Update\GFExperience.NvStreamSrv\x86\server
C:\Users\All Users\NVIDIA Corporation\GeForce Experience\Update\GFExperience.NvStreamSrv\amd64\server

but doesnt work either :(

heydude1337, 2016-10-30:
For me it didnt work either. But renaming the rxinputgamepad.dll in both folders did the trick :)