moonlight-stream/moonlight-embedded #289: Audio skipping issues

shayaantx, 2016-01-05:
Hello, first thanks and props to people who work on this.

Second, I'm currently having an issue every week where audio skips after a 4-5 days of no issues (no video lag or input lag).

I'm running a windows 10 machine with the GFE beta latest, geforce gtx 970 (4mb of ram for gpu), 16gb of ram for the machine. Using the raspberry pi (all drivers and everything is updated). Using latest moonlight (from apt-get sources).

Literally once a week, the audio seems to degrade (very noticeable skips). The video is clear and the input is fine (streaming with 30fps over 1080p). If I bring it down to 720, the audio does not improve. I see the ALSA write errors, I've read in some issues here. However I see those when there is no audio issue, so at this point I can't attribute these errors to my audio problem yet. Everytime this issue happens I usually test running a mp3 on my raspberry pi (outside of moonlight) and I don't ever have issues.

What has consistently (every time) worked for me, is uninstalling the GFE server, and restarting my machine completely, then reinstalling. I won't have an issue for a few days. My computer is really not running anything when I am game streaming (other than GFE and that actual app I'm streaming).

- I should mention I don't really use games or apps via the GFE server config. I just add the mstsc rdp file to the GFE list and just access my computer as a remote pc through the pi (this is perfect for me and I've had no video streaming or input streaming issues whatsoever)

Any suggestions on what I can troubleshoot further?

KeldorDE, 2016-01-06:
I'm noticed the same issue the last days. Sometimes I get the ALSA errors but the audio is still available. Sometimes the Audio is silent for 3 or 4 seconds without any errors. This issue happens after some time and repeats 2 or 3 times, after that the audio resumes normally.

Sounds similiar to your problem. I'm using the latest GFE Release and a RPI 2 as moonlight host (Raspbian Jessie)

shayaantx, 2016-01-06:
I mean it could be network or something, but I've got pretty beastly networking components in my setup.

Got the nighthawk router and a pretty good asus N wifi adapter. I am running my raspberry pi over ethernet and my server is running over wifi. I usually have no issues, can play gta5 or witcher 3 on ultra on 30fps/1080p with no issues, then this audio problem crops up. Literally I can stream witcher 3 on 30fps/1080p while people are streaming netflix and playing xbox one with no issues. So it just seems like networking isn't the issue (yet). 

When these scenarios occur, I usually kick everything off my network except my server and pi, then I downgrade the resolution to 720p, with no success. Up until yesterday reinstalling GFE would fix it, but yesterday the issue occurred in the morning and later at night (mind you I've changed nothing except disable unused sound adapters - as I have a theory that the multiple sound adapters are part of the problem), the issue went away. 

- However this is not consistent with past attempts where reinstalling GFE caused the issue to go away.

diagonali, 2016-01-06:
I'm also experiencing difficulty (as others on other threads have mentioned) with audio stuttering. No idea whether it's Moonlight that can fix it or else the issue lies with my Raspberry Pi, OSMC or possibly Gamestream from Geforce Experience. I do hope this can get looked into and fixed because otherwise it was working ok. Interestingly, it is improved a bit by launching games directly and not through Steam Big Picture.

KeldorDE, 2016-01-07:
I've also experiencing some strange behaviour with the bitrate parameter.
Before it was set to 22000, now I've raised it to 24000 and the sound problem not happens that much as before.

I'm not sure if the problem RPI, GFE or Network related. I've checked all channels when the problem occurs. Checking load on RPI, checking the kern and syslog on the RPI, checking moonlights output, checking network traffic, checking the ping. But all seems ok and I get no errors in any of the checked logs.

KeldorDE, 2016-01-07:
Another thing I noticed is: My RPI 2 has some time shifts. Now the NTP daemon adjust these wrong date/time.
If it possible that the moving time could sometimes force dropping of some network packages like the audio packages? But the only audio is affected, video has no problems.

shayaantx, 2016-01-09:
This issue occurred for the 3rd time to me today. Uninstalling GFE, restarting host machine, install GFE again, repair, audio is flawless again.

- In the first case I reinstalled the latest GFE. In second case, I installed GFE beta, this time, I uninstalled beta and reinstalled the latest non beta GFE (like first time).

- Although the above data would suggest the GFE server is the culprit, but I am stopping the machine for periods at a time, I can't imagine what is actually happening with the server application that is causing the reinstall to fix the issue.

My only thought is to modify the code to dump all the streamed audio data (somehow, don't know much about the code yet) on the pi to verify its not the client that is failing here. Really seems like the server is at fault.

KeldorDE, 2016-01-15:
At this moment I got the Geforce Experience update for Version 2.9.1 and found the following in the changelog:
"Fixes corruption an audio streaming from PC to SHIELD."
I'll test if this is releated to our issue.

shayaantx, 2016-01-16:
I'll try as well.

KeldorDE, 2016-01-16:
For me the issue is still there and not fixed with the ne GFE version :/

diagonali, 2016-01-20:
I've also been having this issue, along with the ALSA write errors and the same uninstalling and reinstalling Geforce Experience *seems* to help. I hope there's a way to get to the bottom of what's causing these issues as they're pretty consistent for people on the Raspberry Pi 1&2. I actually hope it's Moonlight since then it could be fixed directly.

shayaantx, 2016-01-21:
KeldorDE - does it reproduce for you easily?

diagonali - yea it would be nice if the geforce server was a bit more transparent, some logs would be great...I constantly get the ALSA write errors, with and without the sound issue.

I did make a post on nvidia's forums, with no replies, https://forums.geforce.com/default/topic/909962/geforce-experience/audio-skipping-issues/ 

I haven't had the issue for awhile, its intermittent for me, so I'm waiting for it to occur again to try more troubleshooting.

I think your best bet is to buy a cheap android tablet with an hdmi port and usb port and see if issue persists with the android app version of moonlight. I have a shield as well and it doesn't seem to happen there, however, the issue is intermittent, and stopped happening on my pi well before I updated to 2.9.1

KeldorDE, 2016-01-21:
@shayaantx - Yes the first x Minutes run without any issue. After this the skipping starts for 2 seconds, then it continues normally and after a few seconds it happens again. After that it continues without issues for the next 10-20 minutes, then it could happen again.

The period between the issues are not constant so it could be 5 minutes or 20 minutes.

irtimmer, 2016-04-03:
Duplicate of #126

closed 2016-04-03 by irtimmer