moonlight-stream/moonlight-embedded #290: Can't connect to server "my server ip address"

runechronos, 2016-01-06:
Hy,

I don't get what's going on, I installed Moonlight and it was working great. Yesterday no problem, today, i'm trying to connect and no matter which moonlight command i type, i get : "Can't connect to server "my_server_ip" ". I checked my firewall on Windows, i even deactivated it, nothing works, i tried by ethernet/wifi, i don't get what's going on. I'm using **Moonlight 2.1.2** and Geforce Experience last version, and a Raspberry on **Noobs 1.5**.

Here is the content of my /etc/network/interfaces file :  

`auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp  
 
allow-hotplug wlan0
auto wlan0

iface wlan0 inet dhcp
wpa-ssid "MyWifi"
wpa-psk "MyKey"
`

shayaantx, 2016-01-06:
is your GFE server actually running? Try below url on your host machine and see if actually returns anything. Also check the services for the Nvidia geforce experience service, the network streaming service, streaming service, etc.

http://127.0.0.1:47989/serverinfo?uniqueid=1234

runechronos, 2016-01-07:
Thanks for your help, it was indeed the "Service" that was not started on my windows host machine, it went off i don't know why it was set to "automatic" not manual, but that was the problem thanks again.

closed 2016-01-07 by runechronos