moonlight-stream/moonlight-embedded #292: Loss Stats: Transaction Failed: 32

FLo481, 2016-01-11:
Since last Friday (08.01.16) I am getting this error (Loss Stats: Transaction Failed: 32), always after about 1 min streaming. I switched from Raspbian Jessie to Wheezy but nothing changed, tried streaming via W-LAN and LAN, the same problem, I also reinstalled my nvidia driver and all of it's components (the same) and i tried moonlight embedded from the repo and I also installed it manuelly but nothing seems to solve this problem. Any solutions and what causes the error?

forteh, 2016-01-11:
I'm suffering from the same problem, stream drops out after a minute with the "Loss Stats: Transaction failed: 32" error.

I'm streaming over ethernet from an i5 2500k@4.5ghz with a gtx660 onto an rpi B+ arn overclocked to 1000 and gpu to 350 if I remember correctly. Streaming -1080 -30fps.

I've updated raspbian wheezy and uninstalled moonlight to no avail, I've just installed the latest nvidia drivers with a clean installation and still the same error. Altering bitrate or packetsize doesn't affect the crash.

I'm a complete novice at linux, only had the rpi set up for a week and it used to work fine!  I've read through some of the related issues but can;t make head nor tail of it :(

Tinus1018, 2016-01-12:
I got the same issue.

I am on LAN with i5 2500k and gtx 780Ti. I use a raspberry pi 2 with jessie and last version of moonlight installed using repo.

I got this issue recently also. (since one week)

forteh, 2016-01-12:
I've just tried unpairing and re-pairing the pc and rpi and it hasn't solved the issue.

Gehilfe, 2016-01-13:
Same problem here since yesterday (2016-01-09) :(
I already tried to reinstall nvidia drivers but without success.
Running moonlight-embedded on rpi2 (osmc).
It happens exactly one minute after starting moonlight

forteh, 2016-01-14:
I uninstalled gfe today and reinstalled the newest version (released today), re-paired and still exactly the same.
My only thought it that it could be a recent windoze update that has killed it? I'm on windows10 pro here.

Tinus1018, 2016-01-14:
I have Windows 10 also and I also suspect a windows update to be the cause

rasmusbl, 2016-01-14:
I am having the same problem since a few days. LAN,, RPi B+ and Windows 10,

irtimmer, 2016-01-14:
It looks like the connection is for some unknown reason dropped. Although I can't reproduce the issue. So without any more information (Wireshark dumps, Geforce Gamestream log files, other Moonlight output) it's almost impossible for me to debug.

forteh, 2016-01-14:
Where about are the gamestream log files?  I'll dig some out for you if it will help debug.  I'm not familiar with wireshark but willing install to help in anyway I can :)

Tinus1018, 2016-01-14:
I will try to do the same when I can to compare.
If you know the location pc the log files, it will helps (found nothing with quick googleing) 

I am also not used to wireshark, I will try.

Otherwise moonlight doesn't provide other useful infos.
I just to insist the fact that the crash happen  exactly after one minute or streaming

forteh, 2016-01-14:
These are a couple dropbox links to nvidia log files from last night.

https://www.dropbox.com/s/2ijpvlyjd9xv758/NvStreamerCurrent.log?dl=0
from C:\ProgramData\NVIDIA Corporation\NvStream

and
https://www.dropbox.com/s/gckye45ifwkm2qw/steamCurrent.log?dl=0
from C:\ProgramData\NVIDIA Corporation\NvStreamUserAgent

I have no idea if these are useful, hopefully they might help you debug :)

cgutman, 2016-01-14:
The 2 logs seem to indicate that it's still thinks that it's trying to start your game when it times out and closes the connection.

In NvStreamerCurrent.log:
#0(I)[2016-01-13 23:02:43,485]{00000D14} Termination reason = GAME_LAUNCH_FAILED (GameProcessNotFound)

What happens if you use the steps here to stream your whole desktop then start the game from there? https://github.com/moonlight-stream/moonlight-android/wiki/Setup-Guide#using-moonlight-to-stream-your-entire-desktop

forteh, 2016-01-14:
That would make sense, I'll try streaming the desktop (I would probably prefer this to be honest over the steam big picture mode) and see if it still does the same.
I don't use full time AV (unless the micrsoft security essentials acts as AV?) and the bog standard windows10 firewall.

FLo481, 2016-01-14:
Streaming the Desktop , with the mstsc.exe,works without crashes. Thanks for the little tipp ;)

forteh, 2016-01-14:
That's great to hear!  I'll test and confirm later on. :+1:

forteh, 2016-01-14:
Confirmed, that it works streaming the desktop without crashing.
However (call me stupid) the display is showing as larger than my TV (both running 1080p 60hz native) and I'm missing the edges.  What am I doing wrong?

arv187, 2016-01-16:
resuming... what is the solution? close antivirus?

forteh, 2016-01-16:
Can't offer a solution on streaming steam directly as it still drops out after a minute; streaming the Windows desktop works perfectly however. I fixed the overscan issue by adding the overscan_scale=1 and output_scale=1 lines to /boot/config.txt.

arv187, 2016-01-16:
then in raspberry pi I must use vnc or similar?

forteh, 2016-01-16:
In order to stream without the crash I have to set up the mtsc.exe in gfe and stream it as an app from moonlight; assuming this is what you meant by vnc?

My command line is moonlight stream -app Windows -1080 -60fps 192.168.1.6

Nothing stopping you starting steam and then running big picture, that works just fine. Streaming the desktop works better for me as I can use it for solidworks if I need to work from home.

arv187, 2016-01-16:
In raspbian not there mtsc app to see desktop game computer. I can use other client vnc app, how tinny vnc or real vnc, are similars teamviewer.

Moonlight work as cliente? then mtsc will the host and moonlight the client to stream desktop?

I will try set up the mtsc in gfe, thanks by answers!

forteh, 2016-01-16:
That's correct, add mtsc as a program in gfe (I renamed it to Windows for ease of use) and it should work directly from the raspbian terminal window/command line.

donfalcone, 2016-01-16:
Hello i have the same issue. How can i setup the desktop app ?

arv187, 2016-01-16:
Follow this link https://github.com/moonlight-stream/moonlight-android/wiki/Setup-Guide#using-moonlight-to-stream-your-entire-desktop

Gehilfe, 2016-01-16:
Thanks for the help! The tip with mstsc brought me to solve the Steam problem. In GFE you have to add the Steam Client (c:\Program Files (x86)\Steam\Steam.exe) and rename it to what you want (eg. moon). Then start streaming with "moonlight stream -app moon -1080 -60fps -platform pi IP-OF-YOUR-PC" and have fun! In my case, the folder "c:\Program Files (x86)" was hidden in Windows 10

donfalcone, 2016-01-18:
Thanks It help me to run It.

Rajackar, 2016-02-08:
I've got this problem as well. stream is killed after 1 minute. Anyone know of a way to set the timeout to something very high so there's no problem?
I tried launching steam as described. That works but only allows me to use my computer monitor native resolution which doesn't match to my TV. Launching with resolution parameters only results in a black screen. 
When not passing the -app parameter the resolution is ok but then I have the timeout problem.
Hope anyone can help me.

cgutman, 2016-02-08:
This is a problem with GameStream's Steam integration. It attempts to connect for 60 seconds before failing and tearing down the stream. It's happening with the Shield devices too and can probably be fixed only with a GFE update.

https://forums.geforce.com/default/topic/909598/gamestream/gamestream-broken-with-steam-and-big-picture-mode/

Rajackar, 2016-02-09:
Looks like more people are reporting this on both the GFE forums and on the steam forums:
http://steamcommunity.com/groups/SteamClientBeta/discussions/0/458606877316542130/

aikoncwd, 2016-02-09:
I did some research with this issue.

Is not a GFE bug, its not a moonlight bug... it's a Steam bug.
Last month, any user who used the Steam Beta Client had experienced the same bug, dropping from the "beta" client solved the issue. Last week the "beta" update was merged in the official Steam Client, so the issue is now populated to every user :(

You can create a LNK file to Steam BigPicture and add it to the GFE custom games:

> For anyone with this issue there is a working fix which I used successfully last night. Create a custom gamestream entry linking to a shortcut with the destination "steam://open/bigpicture"

It's just a noob solution, but it works. Some users report that the size of Gamestream screen is smaller using this "trick", we can't do nothing more than waits for Steam move. I opened a ticket to the steam support. If we all open tickets, maybe steam will work for a patch/update soon.

arv187, 2016-02-09:
Im from spain too, to non english speakers and people without technical knowledge, can you write the exact issue to open ticket in steam?
Then the issue is from steam or nvidia?

posti85, 2016-02-09:
Well, it seems many of us are from Spain hehe. Has somebody tried to reproduce the issue using moonlight official (PC or android, for example)? Maybe a problem with moonlight embedded in our Raspberry?

aikoncwd, 2016-02-09:
@posti85 I tried, same bug. People with official nVidia Shield deviced are getting the same issue/bug. Steam Client was updated and now BigPicture crash when its invoked by Gamestream. You can invoke gamestream with official nVidia Shields or custom moonlight-embedded (RPi), but you will get the same issue :(

Try to open so many tickets to the steam-support! :)

posti85, 2016-02-09:
Well, so if it occurs  with the official nVidia shield,  I think it is a 'good new', Valve shouldn't let that issue opened much time, because it affects to nVidia directly. Let's see what happen.

coyote4k, 2016-02-09:
yes they will certainly attach developers in the office with chains to work day and night. : )

coyote4k, 2016-02-09:
I transmits details the procedure that aikoncwd talks earlier in the post and sent me and I thank , work great !!!!

For those who want to try, I confirm that it works on the last distri jessie 2016-02-09, and the last GFE 2.9.1.35.

"aikoncwd " > Create a new shotcurt (right click on your desktop, new shortcut) and paste this
steam://open/bigpicture
Save it with a common name like: STEAMLINK
Add this STEAMLINK into your GFE and use this to launch: moonlight stream -app STEAMLINK

I can even tell you I bought a steam link I returned because too much latency can not play fast game (car and fps). Now thanks to aikoncwd I just play a project cars is what is better, it works better than Shield steam after link of what I can see.

Tonight I reinstall my SD with Raspian multiboot + + kodi retropié and steam for test script.

Streaming Steam Great work !!!!!!!

Thanks aikoncwd 

Coyote4k

fl3xxx, 2016-02-10:
@coyote4k awesome! This works for me too! I´m using the last dist of Wheezy and the beta Version 2.10.1.2 GFE. 
Thank you so much!!

coyote4k, 2016-02-10:
Yes works on Wheezy :  My sd card multiboot with Openelec Kodi , Rapsian ,  retropie and moonlight inside work , i make a batch in retropie with different moonlight launch script , pair , 1080 , 720 , 720 30fps , Shortcut-method : )  ect and launch direcly with the game pad and steam work perfecly with shortcut-method .

After play , i quit steam with the gamepad after on my keyboard press touch windows , close gamestream windows and menu retropie return on my TV yeahhhhhhhh !!!!! .  perfect 

Thanks more aikoncwd . 

Coyote4k

Rajackar, 2016-02-12:
The problem with steam seems to be fixed in the latest steam beta.

closed 2016-03-27 by irtimmer