moonlight-stream/moonlight-embedded #295: NvStreamUserAgent.exe crashes with new http pairing

AreaScout, 2016-01-18:
When pairing with latest revision ( http pairing ) NvStreamUserAgent.exe crashes, so pairing never works 

Machine:
Windows 10 64bit
GeForce TI750
GeForce Experience 2.10.0 Beta + 361.43

cgutman, 2016-01-18:
Can you upload the contents of your "C:\ProgramData\NVIDIA Corporation" folder? ProgramData is a hidden folder BTW.

AreaScout, 2016-01-18:
Here you are, i always have to restart 'NVIDIA Streamer Service' that GFE 2.10 accepts connections and if it accepts connections and i start streaming, the GamesStreaming background app ( black GameStreaming screen ) is not starting, it only starts when i manually start 'nvstreamer.exe -sci -shield' then it kills that task, starts it's own nvstreamer.exe and starts the background app ( black GameStreaming screen ) and finally starts the game.    

http://www.areascout.at/GeForce%20Experience.zip

p.s.: i had to pair with an older version of moonlight

AreaScout, 2016-01-18:
Here is a new Log with 3 pairing attempts at the end of the file, i hope GfeClientLog is enough 

http://www.areascout.at/GfeClientLog.zip

AreaScout, 2016-01-20:
NVIDIA support team helped me to track down my connection problems, also the crash when pairing was caused by this, it was a conflict with 3rd party software ( Kaspersky Anti Virus )

Thanks for investigating

AreaScout, 2016-01-29:
Issue still exists, while my connection problems are gone now it seem's that pairing still crashes NvStreamUserAgent.exe, i had i working once :(

AreaScout, 2016-02-27:
I can confirm that this fixes pairing

thank you ! :dancer:

RG

PR #316 (2016-02-28): Fix pairing on v2.1.3
There was a latent stack corruption bug that happened when pairing. It wasn't visible until the UUID stuff got added. As a result, the UUID gets appended to the challenge response output, breaking it and causing pairing to fail every time.

This should fix #295 

Since the current release can't pair at all (on at least the Jessie build), I'd suggest that you make another build after this fix.
closed 2016-02-28 by irtimmer