moonlight-stream/moonlight-embedded #299: Loss Stats: Transaction failed: 104" then crash

Gruncher, 2016-01-22:
I'm running this on a Raspberry Pi 2 and trying to get it going smoothly.

Last night I had it streaming GTA5 at a fairly smooth framerate for about 30 minutes, and I closed it to try another game.

Now I get this error: "Loss Stats: Transaction failed: 104" after about 30-60 seconds of launching a game, and it immediately shuts down and stops streaming.

Specifically, it looks like this:

```
$ moonlight stream -mapping /home/alarm/PS3.map 192.168.1.132 -app witch
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Alsa shortm write, write 16 frames
Alsa error from writei: -11
Alsa shortm write, write 236 frames
Loss Stats: Transaction failed: 104
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
```
Anyone have any clue? I tried restarting both the host and the Pi, it didn't help. No idea what I broke here. :(

Gruncher, 2016-01-22:
Okay, I don't want to celebrate prematurely, but I wiped out my nvidia drivers from the host machine and installed an older (december) version of the driver. I was able to stream a game perfectly for about 10 minutes, so I'm really hoping this fixed the issue.

irtimmer, 2016-02-08:
Duplicate of #201

closed 2016-02-08 by irtimmer