moonlight-stream/moonlight-embedded #301: Crashes

BuzzKilll, 2016-01-26:
Hey, I have just started using moonlight, and yay it works on my Raspberry pi, but, every time i launch steam Via moonlight when i start a game it crashes.

P.S (Once i caught it saying logging off just after i logged in)

srounet, 2016-01-28:
Same problem here.
Built from source on arch and a Raspberry pi. I'm able to see Steam in "Big Picture Mode", but when I launch a game I have the same issue as @BuzzKilll .

```
[root@steampi build]# ./moonlight -60fps stream 192.168.1.16
Moonlight Embedded 2.1.2 (EMBEDDED)
Connect to 192.168.1.16...
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Alsa shortm write, write 16 frames
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
```

Gruncher, 2016-01-29:
Try installing an older version of geforce experience.

posti85, 2016-02-04:
Same problem here. When I stream Steam, it shows the big picture mode in my TV, but after exact 1 minute (no matter if I start a game), it stops and the log contains the same as @srounet reports.

It appeared since the last Steam update, but I'm not sure if it is the cause of the problem. I ended up streaming the whoole desktop (C:\windows\system32\mstsc.exe using 'Desktop' as the moonlight app name) and for now it works. But using Steam gui directly was more comfortable than using the whoole desktop. I hope a solution can be found.

BuzzKilll, 2016-02-05:
@posti85 can you give me the moonlight stream for the desktop, newb at moonlight Thanks in advance

BuzzKilll, 2016-02-05:
@posti85 when i try to do moonlight stream -app desktop 192.168.1.3 it dosent work. any suggestions?

posti85, 2016-02-05:
@BuzzKilll, Desktop app doesn't exits by default. You have to create it. Have a look to the 2 last points in this guide: https://github.com/moonlight-stream/moonlight-android/wiki/Setup-Guide#adding-custom-programs-that-are-not-automatically-found

coyote4k, 2016-02-08:
Hello ,
RPI2 moonlight-embedded with alsa crash

New and old version GFE and with limelight or moonlight result ..... crash .

With all command -30fps -720 -localaudio and more crash

raspi-config Force hdmi crash more

apt-get remove alsa-utils and apt-get install alsa utils > sudo reboot = crash

apt-get install pulseaudio crash

In 2016 impossible work a stream steam big picture on the RPI , you have a solution .

Problème code Alsa.c or another problem maybe firmware RPI , you have a solution ?

info : Moonlight official work perfecly on my smatphone galaxy S2

GFE or steam no in cause . i think firmware RPI2 or alsa.c need verify code and for this i'm not
have a expérience .

Big up for this problem open #126 ^^^^^^^ three night work for run moonlight correclty and are a silly man now : ) .

i start the stream and time bomb it's activated only picture or in game , 25s past Boooooom crash . fucking shit

log :

moonlight stream 192.168.1.2
Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Alsa error from writei: -32
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done

And more 

sudo apt-get remove moonlight-embeded 

and install manualy olf version moonlight 2.0 and GFE 2.1.5.0 .

sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
tar xzf cmake-3.2.3.tar.gz
cd cmake-3.2.3
./configure
sudo make install

Compile and install moonlight:

cd moonlight-embedded-2.0
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE=Release ../
make
sudo make install 

All install ok with no error and what  crach Alsa more fuck 

and look it's possible : https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=65878&sid=0aaab1326ac7b0f69c9551d426df3465&start=250 

Look the finale message : by AikonCWD » Fri Jan 15, 2016 12:02 am
Thanks for your work man! This is how moonlight looks on my RaspberryPi2

https://youtu.be/djBzL_jI1tk

Launched via emulationstation menu, Witcher 3 at 1080 @ 60 :D

https://www.youtube.com/watch?v=djBzL_jI1tk&feature=youtu.be work 

four day and night for this alsa error message anyway , I'm desperate

coyote4k, 2016-02-09:
Alsa shortm write, write 12 frames
Alsa error from writei: -11
Network dropped an entire frame
Waiting for IDR frame
Waiting for IDR frame
IDR frame request sent
Alsa error from writei: -32
Loss Stats: Transaction failed: 32

 Updating firmware
 *** Updating kernel modules
 *** depmod 4.1.17+
 *** depmod 4.1.17-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 0764e7d78d30658f9bfbf40f8023ab95f952bcad

uname -a
Linux raspberrypi 4.1.17-v7+ #834 SMP Mon Feb 1 15:17:54 GMT 2016 armv7l GNU/Linux

Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Alsa shortm write, write 12 frames
Alsa shortm write, write 20 frames
Alsa error from writei: -11
Network dropped an entire frame
IDR frame request sent
Alsa error from writei: -32

[donwngrade) sudo rpi-update 2ef601a50b68eebeeb4dc3c6c525855961891be6 

*** Updating firmware
 *** Updating kernel modules
 *** depmod 3.10.25+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 2ef601a50b68eebeeb4dc3c6c525855961891be6
 *** A reboot is needed to activate the new firmware
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done

black screen Sd out :/ 

http://downloads.raspberrypi.org/raspbian/images/raspbian-2015-02-02/

GFE old and last with moonlight on raspbian-2015-02-02 old image for rpi2 crach

Alsa shortm write, write 24 frames
Alsa shortm write, write 232 frames
Alsa error from writei: -11
Alsa error from writei: -32
Loss Stats: Transaction failed: 32

and with limelight on raspbian-2015-02-02 . 

java -jar limelight.jar stream 192.168.1.2
Starting Steam
Starting RTSP handshake
Starting control connection
Starting video stream
Starting audio stream
Starting input connection
java.net.SocketException: Broken pipe
        at java.net.SocketOutputStream.socketWrite0(Native Method)
        at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:109)
        at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
        at com.limelight.nvstream.control.ControlStream$NvCtlPacket.write(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.sendPacket(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.sendLossStats(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.access$100(Unknown Source)
        at com.limelight.nvstream.control.ControlStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagramSocketImpl.java:143)
        at java.net.DatagramSocket.receive(DatagramSocket.java:781)
        at com.limelight.nvstream.av.video.VideoStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagramSocketImpl.java:143)
        at java.net.DatagramSocket.receive(DatagramSocket.java:781)
        at com.limelight.nvstream.av.audio.AudioStream$1.run(Unknown Source)
Connection Terminated The connection failed unexpectedly

Resume this is a problem with steam send a update with time bombe for disconect , no because galaxy s official moonlight work .

Steam check RPI2 on the update , it's for block for commercial reason Steam link after november ???  

A program block is install on Geforce server , uninstall antivirus , firewall , full open intenet box .

Downgrade distro fev 2015 

and crash more ... 

Fuck Steam

posti85, 2016-02-09:
I was thinking about the same: a problem with Steam. For me, it was working properly until last Steam update. But I don't know if it is a planned crash (well, if it happends after exact one minute, we could think it is...), are there any way to fix it?

irtimmer, 2016-02-09:
Duplicate of #292

closed 2016-02-09 by irtimmer

coyote4k, 2016-02-09:
Posti85 I'll keep you informed : yesterday send email at AikonCWD which for him works . 
I receive this : 

Hi
Let me check if my Moonlight is working well, I didnt used since last week. Also I recommend you to reinstall the whole SD and start a "fresh" installation:

- Download official Raspbian Jessie Lite
- apt-get undate/upgrade/dist-upgrade
- rpi-update
- Edit apt/sources.list
- Install moonlight using apt-get install (http://embedded-software-architecture.com/?page_id=140)
- Try moonlight

But as I said, Let me check if my moonlight is still working, I will writte you again tonight.
Bye!

I'll keep you informed

coyote4k, 2016-02-09:
Confirmed for AikonCWD . 

2 weeks works and now no works because steam update beta .

Source here : https://forums.geforce.com/default/topic/909598/gamestream/gamestream-broken-with-steam-and-big-picture-mode/

for information steam rapport update 2 fevrier 2016 write : update sound 5.1 for home standing . 

Conclusion : Alsa error it's a sound card of raspberry , steam update sound in 5.1 homme standing after 2 fevrier 2016 all no work and return Alsa write error . 

The source of problem is update steam 2 fevrier 2016 . 

Sujet close , i continue on  #292

posti85, 2016-02-09:
Thank you for the research coyote, good job :) we hope the bug to be solved someday.