moonlight-stream/moonlight-embedded #302: Cant Connect To Server X.X.X.X

liquidaussie, 2016-01-29:
I've just set up Moonlight for the first time, had a few issues, but managed to get all the resources installed, but when I run the Moonlight Pair MyIPAddress, it just comes back saying it cant connect.

I've Restarted both the Pi and the Host PC, Flushed and renewed my IP address, and uninstalled and reinstalled GeForce Experience. I'm currently running 2.9.1.35.

I'm at a a loss for a next step

AreaScout, 2016-02-01:
try to use an older moonlight version prior to this two commits e2ac244 3a6e003

let's say:
`
git reset --hard e89eba9
`

rebuild moonlight and try pairing again

liquidaussie, 2016-02-01:
Thanks. Yeah, I flashed a fresh rasbian install on the sd card, reinstalled moonlight, and used Ge force experience 2.7,and it worked.

closed 2016-02-01 by liquidaussie