moonlight-stream/moonlight-embedded #304: unable to use keyboard and mouse

KaeroDot, 2016-01-31:
I have: Raspberry Pi B, steam computer, connected by wired lan.

I can connect by moonlight to a desktop computer and to a Steam application or Skyrim game. Speed of streaming is without any lags or distortions. Unfortunately the mouse and partially the keyboard is not working in Skyrim. Both Steam and Skyrim thinks I am using controller by marking entries in dialog windows by "A", "B" and other controller buttons. Still the Steam can be controlled by the keyboard and by the mouse (two mouse cursors are shown in Steam application, only one is controlled by the mouse), but in Skyrim only arrow keys are working. Skyrim expects controller buttons to be used when playing (e.g. to confirm loading of a saved game position, Skyrim shows me I have to pres "A" to confirm and "B" to leave).

When showing Add/Test Controllers in Steam application, I can see detected controllers "XInput Controller" and "NVIDIA Shield".

To the raspberry Pi only USB mouse, USB keyboard and HDMI is connected. I have tested several usb keyboards and mouses.

To run a moonlight, I use following command:
moonlight stream -app X ip.address
where X is either "Steam" or "The Elder Scrolls V: Skyrim". When Skyrim is run from Steam, the behaviour is identical.

following input devices are present in /dev/input:
event0
event1
mouse0
mice
I tried to add "-input /dev/input/XXX" options when running Steam or Skyrim in any combination of mentioned inputs, but there was no effects. Only for some combinations the streaming crashed when input was used.

When I try to play Skyrim or to use Steam directly on desktop PC, no controller buttons are shown or required in dialogs but normal keyboard keys are required ("Tab", Enter",...).

Did someone had similar problems? I do not have any controller and never used any controller, only mouse and keyboard.

Is there something other I should test or set to resolve the issue?

shayaantx, 2016-02-20:
I believe if I am understanding you correctly, you don't use a controller, yet skyrim thinks you are?

There is an option in the game where you can turn off the controller, go to settings in game, go to gameplay, and turn off the 360 controller option, then it should force to keyboard/mouse. You can still use the arrows to get to that part of the menu or do it outside of the game stream.

irtimmer, 2016-04-03:
Extra controllers are provided by Gamestream for passtrough support and is not a bug in Moonlight Embedded.

closed 2016-04-03 by irtimmer

KaeroDot, 2016-04-03:
shayaantx: after two months I had finally chance to test it - yes, in Skyrim is a 360 controller option. I switched it off at desktop computer, than I started the game again by means of moonlight, unfortunately Skyrim again thinks I have controller (and the escape function used to show Skyrim settings do not work so I cannot change the settings).

 irtimmer: do I understand you correctly that Gamestream always adds some virtual controller, thus Skyrim started through moonlight always find some controller connected?
Do you think there is any possibility to use keyboard with Skyrim started through moonlight?

irtimmer, 2016-04-03:
You can try to remove the support for Gamestream virtual controller completely by removing a dll. See https://github.com/moonlight-stream/moonlight-android/wiki/Setup-Guide#using-a-gamepad-connected-to-the-pc-instead-of-the-streaming-device

KaeroDot, 2016-04-06:
irtimmer: yes, this works! thanks! renamed file, restarted PC (just for sure), started streaming and skyrim immediately showed mouse cursor and keyboard is also working. in the skyrim settings, the 360 controller is checked, however this doesn't impact the use of mouse/keyboard.