moonlight-stream/moonlight-embedded #305: Stream just stops

cqwer, 2016-02-03:
Hey, i installed Moonlight two days ago on a complete fresh Raspbian Jessy OS. Now i tried to get streaming to work, i tried to stream Lego the Hobbit, it worked quiet well, but after about 2 minutes, it stopped, just like if i pressed shift + alt + crtl + q. No Error or something, streaming PC was just on the desktop. Can anyone tell me what is happening there? 

Another problem is: two days ago, i could play "The witness" for about 10 Minutes without any problem. This game is shared to me over family sharing from a friend. Now everytime (and only then) when i try to start it over moonlight, steam tells me, i have to ask for authorisation to play it. Is anyone else having such problems?

RikJm, 2016-02-09:
I have been experiencing this issue lately, after roughly 2 min my stream will stop and the pc gets left on desktop. I seriously doubt this is a moonlight issue, after a few tests with a few nvidia driver's versions and steam (Beta and standard) it looks like this last one is the one crashing.

I'm going to test it with a DRM-free game tonight and I'll know for sure, but I just experienced this issue after I upgraded to windows10 which made me think it had something to do with it, however it looks like windows10 and gamestream on moonlight were working before the latest steam updates, so it all points down to Steam being the issue here.

cgutman, 2016-02-09:
The issue here is an incompatibility between the latest Steam update and GFE's Steam integration. See this thread: https://forums.geforce.com/default/topic/909598/gamestream/gamestream-broken-with-steam-and-big-picture-mode/

RikJm, 2016-02-09:
Exactly, def not a moonlight issue. 

Now the beta that was causing these issues got pushed to RC with the bug in it. There is a fix, I read it here: [https://forums.geforce.com/default/topic/845023/gamestream/can-t-find-help-for-this-error-code-80030003-/post/4790233/#4790233](url) 

I've tried it and it seems to hold, I haven't played much since I'm a bit busy tonight but it seemed to be working fine.

> DigitalGriffin said:Ok I think I found the solution. Source(https://www.reddit.com/r/nvidia/comments/2fpl3q/steam_not_working_on_gamestream_on_the_shield/)
> 
>     1. Get on your pc
>     2. Right click your desktop, click new, click shortcut
>     3. In the box that says type the location of your item: copy and paste this "steam://open/bigpicture"
>     4. Name it whatever. click finish
>     5. Open Nvidia Gforce Experience also on your pc
>     6. Click preferences
>     7. Click Shield
>     8. Click the plus sign to add new game and add the shortcut on your desktop that you created

aikoncwd, 2016-02-09:
Duplicated -> #292

closed 2016-02-09 by irtimmer