moonlight-stream/moonlight-embedded #309: Can't connect

PrototypeXenon, 2016-02-14:
Hello, I've tried installing moonlight twice now. Both times, I get can't connect to server when I try to run "moonlight pair [My ip]" I've made sure GFE is open, I've turned off AP isolation, I've made sure both devices are connected to the internet/ I believe the issue may have something to do with the fact the the PC is using wifi and the pi is on ethernet. Do you guys have any possible fixes?

PrototypeXenon, 2016-02-14:
Update: I managed to run ethernet to the PC, still doesnt work

gethigh38, 2016-02-14:
Which version of GFE you using ? try to update to the last version (2.9.1.35) as the previous one removed the Shield menu. and make sure you add the Steam app in the shield settings.

PrototypeXenon, 2016-02-16:
I'm on the latest version. Will not having steam added not allow it to pair?

shayaantx, 2016-02-20:
Steam is added by default I think in GFE and no that doesn't matter.

Check your nvidia runninng services, which ones are running and which ones are topped?

Have you tried pinging your GFE server from your moonlight machine? If so whats the result?
What happens on the GFE server when you go to this url? http://127.0.0.1:47989/serverinfo?uniqueid=1234

PrototypeXenon, 2016-02-20:
The two devices can't ping each other, but that link prints out some XML, including my PC specs, so that must be working. My running nvidia services are:

Display Driver Service
Streamer Service
Streamer Network Service
Network Service
Geforce Experience Service

All of which are running

shayaantx, 2016-03-07:
If your your pi and your host machine cannot ping each other, then something in your network or on your host machine is preventing that access.

- Your network/router could be blocking icmp requests (ping for example).
- Your host machines firewall may need to be changed to allow incoming/outgoing access to the GFE network service.
- I don't know what kind of network setup you got, so I can't say for sure.

Once you are able to ping from both machines, you will most likely be able to continue with the pairing/streaming process.

Tjda, 2016-04-06:
Do you have space on the microSD card? 

I had the same problem, because I was not using all the microSD card on Raspbian.

sudo raspi-config / ans choose enlarge

closed 2017-03-18 by irtimmer