moonlight-stream/moonlight-embedded #310: Fails when pairing URRRGENT

OllieChaos, 2016-02-14:
Ok, so I am not that much of a tech genius, but I managed to get to the stage where I have to pair, and when I type in moonlight pair [my pc's ip], it says the 4 number code, and says successfully paired, but nothing comes up on my pc, I have GeForce Experience running on the SHIELD screen when I am logged in. I have tried reinstalling GFE to an older version, and rebooting. I kinda need this sorted quickly because I am making an arcade cabinet in school and I need it to work so I can put my pi inside.
btw I'm running windows 10 on my main pc, if that helps

shayaantx, 2016-02-20:
Moonlight says "Successfully paired" but nothing comes up on your GFE PC?

You should of got the 4 digit code from your moonlight machine, then entered this code on your GFE server/pc. Were you presented with this option on your GFE pc?

OllieChaos, 2016-02-23:
it didn't come up with anything on my pc, so I couldn't type anything into it :(

steve00, 2016-03-04:
I also have this issue.On moonlight embedded 2.1.4 and GFE 2.5.12.11

dead, 2016-03-04:
GFE 2.5.12.11 is beta and its not supported yet.

OllieChaos, 2016-03-05:
I tried using current and older versions, and none of them work

shayaantx, 2016-03-07:
"it didn't come up with anything on my pc, so I couldn't type anything into it :("

- So you are getting "Successfully paired", yet you are not actually entering the code on your host machine? That doesn't seem right and you shouldn't get that unless you entered the code on the host machine. What is the exact command you are running?

- Can your pi ping your host machine? Can your host machine ping your pi?

- On your host machine, do you get results for either or the below URL's?
http://127.0.0.1:47989/serverinfo?uniqueid=1234
https://127.0.0.1:47984/serverinfo?uniqueid=1234

Like I mentioned already, you type moonlight pair IP, this gives you a code on the pi, if your GFE server is active/running and your pi can actually access your GFE server, your host machine will show a popup from GFE server. The popup is where you enter the code. Finally the pi will print out "Succesfully paired".

If you are getting "Succesfully paired" and not entering the code, something is wrong with your setup or install, somewhere....

OllieChaos, 2016-03-07:
OK, thanks for the reply, for the moment I've decided to switch to emulationstation until I finish my project, but I may try switching back to Moonlight. Thanks for all the help :)

irtimmer, 2016-04-03:
Pairing issues should be resolved with latest release.

closed 2016-04-03 by irtimmer