moonlight-stream/moonlight-embedded #313: Starting control stream...failed

on4r, 2016-02-19:
I followed [this](http://jordanhotmann.com/guide-install-moonlight-on-raspbianretropie/) guide to setup Moonlight with RetroPie.
Everything works fine until i try to `moonlight stream` 

Whatever options i tried, i always get the same output:
```
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 0
failed: -1
```

My main PC starts to show Steam or whichever Game i specified via `-app`
but the RaspberryPi stops working after `failed:-1`.
I connect via ssh and have to _ctrl-c_ out. After this the RPi starts to respond again.

**My Setup:**
RaspberryPi1 B+
RetroPie 3.5 
Moonlight-Embedded 2.1.2
GFE 2.10.2.40
Driver 361.91

I tried everything provided in [this](https://github.com/moonlight-stream/moonlight-android/wiki/Troubleshooting#no-video-displayed-on-device) Troubleshooting and couldn't find any threads about this `Starting control stream...Start B failed: 0` error, so i'm hoping for some help here.

irtimmer, 2016-02-19:
GFE 2.10.2.40 has some changes in the protocol and is therefore not yet supported

on4r, 2016-02-19:
Thanks for the fast reply. 
Seems like i have to wait for a new release or install some older GFE version.

Any work in progress status?

irtimmer, 2016-02-19:
First patches already landed in master :P

on4r, 2016-02-19:
Nice! Keeping an Eye on.

irtimmer, 2016-02-19:
New version with the needed patch has been released

closed 2016-02-19 by irtimmer

on4r, 2016-02-20:
Still getting the `Starting control stream...Start B failed ` error but this time with another Error Code.
```
Moonlight Embedded 2.1.3 (EMBEDDED;PULSE;PI)
Connect to 192.168.178.124...
Stream 1280 x 720, 30 fps, 5000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 2
failed: 2
```

on4r, 2016-02-20:
After some digging i'm definitely a little bit smarter but there are still many things i don't understand.
***
You define `LastSocketError()` which uses the `WSAGetLastError()` function to return an Error Code.
Possible Error Codes for `WSAGetLastError()` range from [10004-11004](http://www.jasinskionline.com/windowsapi/ref/other/winsockerror.html). 

But my errors are 2 and 0.

I found that `WSAGetLastError()` is a sort of extension of `GetLastError()` which can report also
windows errors.
So, a return value of 2 can be a [valid value](https://msdn.microsoft.com/en-us/library/windows/desktop/ms681382(v=vs.85).aspx).
 
`v2.1.2`: **0**  ERROR_SUCCESS (The operation completed successfully)
`v2.1.3`: **2** ERROR_FILE_NOT_FOUND (The system cannot find the file specified)

I didn't change a thing between these two versions. 

Nevertheless, the main Problem seems to be an empty reply from [sendMessageAndDiscardReply()](https://github.com/moonlight-stream/moonlight-common-c/blob/master/limelight-common/ControlStream.c#L276-L286)

And because [Send START A](https://github.com/moonlight-stream/moonlight-common-c/blob/master/limelight-common/ControlStream.c#L469-L475) gets called without exception there might be some problem with the [startBGen5[]](https://github.com/moonlight-stream/moonlight-common-c/blob/master/limelight-common/ControlStream.c#L68) Array.
***
I'm no C-Developer and don't really understand what's happening in this Code but i'm very passionate about this project and would love to stream some games on my TV in the future!
Any help appreciated!

Athou, 2016-02-25:
Same issue over here, worked fine with GFE 2.9:

```
Moonlight Embedded 2.1.3 (EMBEDDED;PULSE;CEC;PI;IMX)
Connect to 192.168.0.102...
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 2
failed: 2
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
```

cgutman, 2016-02-25:
Are these self-built binaries or the official 2.1.3 builds? What version of GFE are you running? On the PC, can you run Wireshark, start a capture, try to start a stream, then save and upload the pcapng file?

on4r, 2016-02-26:
Re-installed everything today but with one change:
The first time i used the Release Candidate (3.5.0-rc3) Version of [CMake](https://cmake.org/download/). 
Now i installed the Latest Release (3.4.3), built Moonlight 2.1.3 and ... it works!

Athou, 2016-02-26:
I'm using the official moonlight 2.1.3 build and GFE 2.10.2.40.

Athou, 2016-02-28:
Nevermind, found the issue, I upgraded the binary to 2.1.3 but forgot to upgrade the libraries (shared objects), sorry.

graphicsxp, 2019-02-08:
same here with 3.16.0.140