moonlight-stream/moonlight-embedded #314: Latest GFE prevents RTSP

djrockgt, 2016-02-21:
The latest GFE 2.11.x.x messes with the RTSP handshake. I cannot get past RTSP after pairing. Had to roll GFE back to 2.10.2.40. Just a heads up.

mikeevans82, 2016-02-25:
Same thing happened to me.  Here is my setup:

Client
  RasPi 2  
  Raspian jessie with apt current to 2/24/2016
  moonlight from repository. Version 2.1.3

Host
  Win 10 64bit
  GTX 970
  Driver Version 361.91
  GFE version 2.11.0.54

command:  sudo moonlight stream

Notes:
 > Steam starts up in big picture mode but moonlight hangs on RTSP.
 > I tried the latest display driver uninstaller in safe mode.  Everything worked until enabling beta GFE.

cgutman, 2016-02-25:
GFE Beta 2.11 isn't supported yet.

djrockgt, 2016-02-25:
I rolled back to 2.10. Ran UDD to do remove the drivers and reinstalled. Still unable to pull the games list. Here's my setup:

Client 
- RasPi 2 
- Latest OSMC/Kodi Isengard 15.2
- Installed moonlight using these instructions: https://discourse.osmc.tv/t/howto-moonlight-embedded-former-limelight/5848
- Luna addon to automate moonlight commands (can't pull a game list from SSH either so it's not Lina issues).

Host 
- Win 10 x64 
- GTX 760 
- GFE 2.10.2.40 

I can provide additional info once I'm back home.

Spamm00r, 2016-02-26:
The same happened with me. 

Everything was running fine until last week, when I updated the nvidia Drivers.

Now only moonlight Iphone app works but not moonlight on osmc and raspberry. After trying to stream a game, the screen just remains black for a minute or so and then Kodi restarts.

I have also tried to revert back to the last three older GFE Versions, but that does not work either.

Maybe it has something to do with the Nvidia Driver aswell.

Strange thing is that the Iphone app works, and I have not updated it.

mikeevans82, 2016-02-26:
I just want to add that I am able to roll back and everything is working.  Take note that I start moonlight using an ssh script, not a gui (never used OSMC).  Not sure if that matters.  Here is my exact process for getting it working:

1. Download nvidia driver version 361.91
2. Run UDD and allow it to work in safe mode.  This is essential.
3. Install the nvidia driver.  I always choose "advanced" and check "clean install"

Works fine after that.   To break it I enable beta GFE.  

Moonlight is at version 2.1.3
I'm using the GFE that installs with driver 361.91. 

THANKS cgutman AND irtimmer FOR ALL YOUR WORK ON THIS PROJECT!!!

djrockgt, 2016-02-26:
I'm going to try all those steps as soon as I get home to confirm that they can reproduce the same results. Should have a full report sometime tonight or tomorrow. I tried ssh and had the same failures so it's not necessarily anything with OSMC's or the Luna addon's scripts yet.

djrockgt, 2016-02-27:
Mike, that did it! The steps worked flawlessly! Was able to pair and pull the games list inmediately. Now the only thing I want to see is if the Windows Update to the nVidia adapters break anything. They shouldn't be anything beta, just trying it out of curiosity in case anyone else follows your steps but wonders what the subsequent updates do after DDU.

The updates are as follows:
nVidia - Graphics Adapter WDDM1.1 
Graphics Adapter WDDM1.2 
Graphics Adapter WDDM1.3 
Graphics Adapter WDDM2.0 
Other hardware - NVIDIA GeForce GTX 760

closed 2016-02-27 by djrockgt