moonlight-stream/moonlight-embedded #315: RPi2 + ME 2.1.3 + GFE 2.10.2.40: Video stucks after a certain time.

lexa2, 2016-02-21:
I have a RPi2 running OSMC and from time to time I do use it as a SHIELD client.
Moonlingt had been installed from the binary packages provided by the "official" repos @itimmer.nl.

There are two issues with moonlight I experience, one major and one minor:

**Minor:** specifying any bitrate higher than 10000Kbps results in video stream hangs for dozens of seconds and moonlight displaying messages like this in the console:
```
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
```

Bitrates lower than 10000 seems to work fine. I've got a Gigabit ethernet here and iperf shows ~94MBps bandwidth with default TCP windows in both directions (to/from RPi2), so I don't think LAN bandwidth is a problem here. Tried to overclock H.264 part of the RPi2 to 333MHz but it don't seem to result in any improvements. Problem is the same no matter I use 720/1080 or 30/60 FPS - as soon as I specify bandwidth higher than 10000Kbps - there are visible problems with decoding (hiccups during any motion-intense scene). The higher bitrate I set - the bigger problems get, up to a point where decoding seems to stuck with nothing changing on screen for a period lasting up to half a minute. Then it starts to work again for some time with the same hiccups and then got stuck again and so on.

**Major:** Having bitrate set lower than 10000 everything seems to work smooth and fine for some time, but after about 30-60 minutes of the gameplay decoding suddenly hangs and never comes back. Input stream and audio streams remain working: game sounds are fine and I do see in-game characters moving on the host PC in response to controller events. Alt-Tabing from the game app which normally should result in pausing the streaming and moonlight process quiting gracefully results in moonlight hang at this point:
```
Discarding RTP packet queued for too long
Network dropped end of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...
^C
```

Ctrl+C at the end is when I got tired of waiting moonlight to exit gracefully and tried to close it using usual *nix keycombo. After Ctrl+C press moonlight process terminated and I got back into bash session. Suspecting that this might be a hardware hang I was expecting that I would have to reset RPi2 to get videodecoding working but that wasn't a case: I tried to start up moonlight instantly and it got working flawlessly for another ~60 minutes. And then I got the same decoding video stream hang again. Restarted moonlight - and got another ~50 minutes of flawless gameplay.

Inspecting logs I've been able to catch a SIGSEGV due to unaligned access in dmesg once but I wasn't able to reproduce it and unfortunately hadn't copied the original error message from the dmesg. Would keep and eye on this though and would update this report in case I'd get that error message again.

Giving that the hangs happened at a interval of time close to an hour I had been suspecting cron or systemd timers to interfere with the moonlight but it had proven to be wrong: I have no cron daemon installed on this RPi2 and the only systemd timer I have is a default timer to clean up temp files once a day.

P.S. OSMC installation had been up to date (apt-get dist-upgrade in cycle until no packages get updated) as of the time of this posting.

substring, 2016-02-25:
Without having troubleshoot as much as @lexa2, i have the same problem after a much shorter playing time. The problems reproduces100% on the intro video of Ultra Street Fighter 4, after a few seconds. You can play the game if you skip the video sequence.
When the video is stuck, you can still hear the sound or use the inputs, even quit the game on PC, but then moonlight gets stuck :
```
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...
```

GFE version 2.10.2.40, moonlight 2.1.3, running 720p@60fps

grad-michal, 2016-02-25:
Same config and simillar problem (even lower bitrate - 8000). I check this also on fresh rasbian and overclocked pi to 1000 clock and gpu to 500.

I also noticed (I will confirmed that tomorrow) that when I set higher bitrate the cpu usage on RP is almost 0. So it looks like the raspberry pi is not getting enough data to work with.

HiTmAn, 2016-02-26:
I'm having the same issue too, e.g. with Darksiders 2 running from Steam Big Picture and other games too.

GFE version 2.10.2.40, moonlight 2.1.3 (compiled myself).

grad-michal, 2016-02-26:
**Hardware**
> 
1. Host hardware - GTX 970, i7 6700k@4500, GFE 2.10.2.40
2. Client hardware - RP2 overclocked cpu to 1000 and gpu to 500 (force_turbo=1), latest moonlight-embedded from repo
3. Connection hardware - DD-WRT router, 1000Mb/s (using hardware switch - LAN<->LAN, so no routing), iperf between host and client stable 98Mbit/s, latency aroung 0.310ms

**Test**
I run test using mtsc.exe to prevent Steam issue and measure bandwith on RP2, check UDP package id and checksum using tcpdump. After connection is set up bandwidth drop to around 200 kbits/s. Please correct me if I am wrong but I believe that this is ok because of steady backgroud of my desktop (no dynamic changes). So to mitigate this I just grab notepad window (white backgroud instead of black backgroud of superputty console) and move it arround to simulate heavy changes. I tested everything with -1080, -60fps and -packetsize 1400 (just below my mtu).

**Observation**
1. Even with bitrate=5000 I see some warning (like: Network dropped middle of a frame, Discarding RTP packet after queue overgrowth) on first seconds of connection
2. After initial connection on 5000 bitrate there is no warnings and everything is working as expected (even when I simulate heavy changes which cause network usage goes up to almost 5Mbit/s)
3. Same results when I reach bitrate 10000
4. Between bitrate 10000-15000 I see some warning in logs about frames being dropped.
5. After 15000 I see constant burst of warning and no streaming at all

Any idea what should I check next?

lexa2, 2016-02-27:
Guys, let me to remind you that the major problem this bug is about is streaming coming to a complete video stream hang after a certain period of time. It's not the bitrate issue that stops me from using RPi2 as a streaming client and being sort-of content with the achieved result. I can more or less tolerate the quality of the stream I get at 8-10Kbps fot 1080p but I cannot tolerate the need to interrupt my gameplay session every ~40 minutes and ssh into RPi2 to killall moonlight & restart streaming.

Reading comments I see that people report flawless behavior with GFE 2.9.1 so what I'm thinking about doing is to try to downgrade from 2.10 to 2.9 and check if both issues would be fixed with that.

grad-michal, 2016-02-27:
You are right and I apologize for off topic. I want to share just one more thing. 

Finally I found the cause of my issue. My host connection to router was not that stable (despite it is negotiated to 1Gbit/s it cannot sustain that speed. This caused massive UPD package drop (more than could be queued in moonlight). 

The easy way to confirm if this is the reason of messages like "Discarding RTP packet queued for too long" is to install iperf3 on client machine (RP in my case) and invoke test for UDP (most test I have done before was for TCP and they actually show nothing interesting because of TCP retransmission) from host to client with this command: 
`iperf3.exe -u -l 1024 -b 20M -c MOONLIGHT_HOST_IP`

This will send UPD packages at 20M bitrate (default for -1080 -60fps flags for moonlight) with default moonlight package size. For local wired network there should be NONE lost datagrams (or only a few like 0,01%). 

For example after I force connection on host machine to 100Mbit/s not a single datagram is lost even with bitrate -b 100M.

cgutman, 2016-02-28:
In GFE 2.10, NVIDIA is doing resolution changes on the fly (as they used to) but this time I could not find a way to turn them off. I think when GFE decides to lower the resolution, the stream hangs. Perhaps there's a config changed OMX event that needs to be consumed?

lexa2, 2016-02-28:
@grad-michal Thanks for suggesting testing UDP loss rate, I was too used to good quality network hardware running at 10GBE or 56IB so hand't considered the possibility of SOHO switch ports queue overflow when transmitting from 1GBE to 100Mbps. Would test it this evening using iperf2. Also would try to force 100Mbps on the switch port facing host PC. In case it would help this might be a reason to consider installing another cheap ethernet card (or use a cheap USB2.0 one) and force streaming traffic through it.

BTW, I have a geniune shield portable console and it seems to work better over WiFi (802.11n in 5GHz band) comparing to wired ethernet connection using Y-cable OTG and USB2.0 ethernet adapter. Built-in nVIDIA's network test measures only about 6Mbps bandwidth over wired connection that is way low and I've been thinking that this is a cheap china USB-2-ethernet adapter issue. Having similar problems with wired RPi2 connection + moonlight makes me suspicious that this really might be a port queue overflow and UDP packets loss issue. Will test.

grad-michal, 2016-02-28:
You should propably try to test it using iperf3 too. Using iperf2 in my case does not showing packet losses, but iperf3 do. I looked briefly at the source code and I believe iperf 2 is writing to socket with some calculated delay after each write but iperf 3 is not.

irtimmer, 2016-02-28:
Fixed with the referenced patch being merged.

closed 2016-02-28 by irtimmer

substring, 2016-02-29:
@irtimmer knowing this is a kind of major issue with GFE2.10, do you plan a 2.1.4 release ?

lexa2, 2016-02-29:
@irtimmer Thanks for clarification and your hard work on the project. Looking forward to download updated binaries (compiling stuff directly on the RPi and even on RPi2 is a painful experience on the one hand and I don't want to spend a lot of time setting up cross-compile environment on the other hand) and would stuck with GFE 2.9.1 for now.

@grad-michal Yep, "minor" problem tracked down to be a poor quality implementation of GFE essentially generating extreme bursts of UDP traffic at the rate matching the host network adapter could handle and then sleeping for some time. While on average the bandwidth seems to be in reasonable 10-30Mbps range the bursts could be something like 512KB of UPD data transmitted at a 200-300Mbps rate. It results in target port queue overflow and huge packet loss. Switching host PC port to 100Mbps fixed the problem. Another way to solve it was to reconfigure switch where I have Shield Portable and RPi2 connected to in a way so target port sits in separate vlan and linux software bridge is used for L2 switching instead of hardware switch chip. My ASUS RT-N16 is running OpenWRT and its CPU is able to handle around 300Mbps traffic passing through software bridge and I've got total control over downstream port packet queue. So now I have host PC port set back to 1Gbps and still could stream over ethernet with adequate bitrate without hitting the "minor" issue. Thanks again for setting me on track concerning UDP flooding burst resulting in packet loss route.

HiTmAn, 2016-03-01:
@lexa2 Why don't you just try to compile from the source? At first I was confused about all this stuff, but it's not that hard as it seems. There's even a handy guide for this: https://github.com/irtimmer/moonlight-embedded/wiki/Compilation
Also if you use the latest Raspbian Jessie, you don't even need to mess around with compiling the development tools from source. They are all now in the repository in apt-get.

lexa2, 2016-03-06:
@HiTmAn Being an HPC unix system engineer compiling from source is not something that frightens me much ;-). Still, compiling for RPi2 requires having cross-toolchain up and running (in case you want things to be done fast; compiling something big like linux kernel on RPi or on RPi2 is really a painfully slow experience, been there and don't want to come back) and that is not something I really want to spend my free time on. I've got enough of compiling and maintenance during work hours so for home PCs I prefer to use pre-compiled binary packages. Luckily it usually doesn't take long for upstream maintainers to produce a new binary package so it's not really a problem.