moonlight-stream/moonlight-embedded #317: Can you add QUIT command in the .map file?

keem85, 2016-02-25:
I have been searching for an answer to this for weeks.. Moonlight never quits when I quit a game/application.. For instance, I'm streaming Dolphin Emulator perfectly, and I have even created a nice little Dolphin menu on my RetroPie.. I have assigned "quit dolphin" as a shortcut on my game controller itself, so I can quit the emulator (or any other game I can click exit).. But the moonlight keeps streaming! You get the "nvidia" logo behind, because Moonlight hasn't detected that I have quitted a game....

"ctrl + alt + shift + q" would not work, because I have no keyboard connected to my Raspberry Pi 2.. But can I assign this combination, or better yet, can I assign "moonlight quit IP" to a button in the .map file? Is that even possible?

HiTmAn, 2016-02-26:
First thing you need to add this to your sourcecode:
https://github.com/runoshun/moonlight-embedded/commit/581715971344f97ed8d27070583fedc3366a1532

It allows you to quit moonlight via gamepad with left and right trigger, left and right button and the ps button pressed simultaneously.

I too have a menu in RetroPie, not for Dolphin but for Moonlight and some games, and if you start them via sh scripts like I, you can just add a script with "moonlight quit" and name it e.g. "00_Quit_GameStream.sh", and it will show up first in the list.

So after you quit via gamepad, just select the "00 Quit GameStream" and it will quit Nvidia GameStream.

Dustee, 2016-02-26:
I have a similar setup with you guys.
What i do to solve this problem is use an app called "RasPi Check" on my android phone.
This app is free and allows you to store ssh commands.
"moonlight quit 'ip address'" does the trick.

As for the dolphin exit hotkey, go to "hotkey settings" on dolphin and bind "Stop" to whatever you like ("Exit" hotkey doesn't seem to work for me).

keem85, 2016-02-26:
Hi HiTmAn!

I use Ishiiruka Dolphin version. It's a lot better than the stock Dolphin, as they have added shaders, sso, better performance all over etc.. Just google it. Ishiiruka has hotkey assignment for gamepad. Its wonderful!

About the source code. Where do I put it? Is it in the .map file? Also, the last tip you gave me, that would imply that I have to actually see the retropie menu? Because when I wuit dolphin, the nvidia shield background (moonlight streaming) is in the way. There is no way to actually see the .sh script I made..

Dustee, I'm doing it the same way you do. But I want it more like HiTmAn tells me.. I want to be able to quit the moonlight stream from my own DS3 controller.

pcca-matrix, 2017-02-01:
What i use for retropie is runcommand for lauching .sh 
put all your .sh in rom folder "steam"
create folder inside configs named "steam" and add emulators.cfg , write
steam = "%ROM%"
default = "steam"

on es_systems.cfg add
  <system>
  <fullname>Steam</fullname>
  <name>Steam</name>
  <path>~/RetroPie/roms/steam</path>
  <extension>.sh .SH</extension>
  <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ steam %ROM%</command>
  <platform>steam</platform>
  <theme>steam</theme>
</system>

and simply add to /opt/retropie/configs/all/runcommand-onend.sh eand write 
moonlight quit IP

irtimmer, 2017-06-14:
With commit cdca4c1b3a3453509af32d20edac1a6eefb4be9a a button combo (Play + Back + Right Shoulder + Left Shoulder) have been added for gamepads to stop the stream.

closed 2017-06-14 by irtimmer