moonlight-stream/moonlight-embedded #318: Xbox controller working in game but not in Steam big picture

Savage87, 2016-02-25:
I seem to be having the opposite problem many are experiencing. I am using a wireless xbox 360 controller with a Pi. I suspect I have a knockoff wireless receiver as I was having issues with the default drivers. I followed a guide to use xboxdrv instead with a custom mapping and that works great while in-game but not in the steam interface.
This seems suspiciously related to the problem where people with controllers attached to the host were having their controller be disabled by gamestream to be supplanted by a controller on the client (fixed by removing rxinput.dll) but in reverse. My controller (on the client) is only picked up after a game launches so I suspect that prior to that in steam big picture, my controller input is not being sent to the host. Does that make sense? If so, any idea what I need to do to fix this?

I am on the latest Moonlight, GE, and steam. I set this all up yesterday on a pi that already had rasbian Wheezy on it. I am having no other issues other than the controller not working in Steam. It is picked up in all other games. The controller is wireless connected to the pi by way of a likely knock-off xbox wireless receiver. 

To set up I run xboxdrv in silent mode and am told the controller is available as /dev/input/js0 and /dev/input/event0 then I start moonlight with 'moonlight stream -1080 -30fps -app Steam -mapping xbox.map -input /dev/input/event0 192.168.2.xx.'

lexa2, 2016-02-26:
If your controller attached to Pi works fine in game but not in Steam it does not make sense to assume that input events are sent to a game and are not sent to the Steam at the same time.

I've had the same problem with controller working in one app and refusing to work in another due to having an F710 receiver connected to host PC. Disconnecting receiver from the host fixed the problem

Another issue I've been having was controller working OK in Steam right after I start up the streaming but stopping to work after I start up a game and quit it (it was The Cave in my case, but YMMV). This issue had been fixed by switching into using beta version of the Steam. Later on - after discovering controller issue due to having F710 receiver connected to the host PC and removing it - I had tried to roll back into mainline Steam build and found out that disconnecting F710 receiver from host PC also fixed this issue with Steam.

To summarize: first of all make sure you do not have any controllers attached to host PC and that you do not have any xinput emulator laying around somewhere beneath game files. Next thing to check is to attach keyboard to your Pi and navigate into Steam settings and check what's up with gamepads Steam was able to detect. Chances are your issue might be related to the incorrect configuration of the Steam.

irtimmer, 2016-04-03:
Issue with Steam and not with Moonlight Embedded

closed 2016-04-03 by irtimmer

ShayBox, 2017-02-11:
Issue with Moonlight, not with Steam. The ONLY game that detects my controller is Binding of Isaac, and even then its buggy. VERY Bad. support more than Xinput.

7hunderbug, 2017-02-12:
@ShayBox The issue is *not* with Moonlight as many people (100s, 1000s?) are using Moonlight for Pi happily and the way it was intended (including me). The only logical explanation then is that there is an issue with your setup and the most likely culprits are Steam and / or GFE or an error in your Pi + OS + controller configuration.