moonlight-stream/moonlight-embedded #320: Video stream freeze in XCOM 2

mtlballer, 2016-02-28:
Every time I try to go to the Geoscape in Xcom 2 the video stream freezes, while audio continues and the game continues on the host pc as well. Freezes every time without fail. Running on RPi 2 model B fresh Raspbian jessie install, latest version of moonlight. Please let me know of any more info I can provide.

lexa2, 2016-02-28:
Had you tried to specify low bandwidth for your stream? What you get might me the same problem(s) as in #315.

mtlballer, 2016-02-28:
Just did some playing around with the bitrate as you suggested, funny enough it seems to be the exact opposite as the problem in #315 as bitrates lower than and equal to 10000 kbps caused it to crash, while a higher bitrate of 15000 kbps let it run. I have no tested it beyond just the sequence that crashes, will do that when I have time later today.

cgutman, 2016-02-28:
@irtimmer This bug will affect any game that has enough action/detail to require significantly more than 10 Mbps to encode (aka most games at some point). GFE will change resolution on the fly to match the bitrate until the video complexity decreases where it will return to the desired resolution. I used to turn this feature off on previous builds of GFE but I haven't found a way to do it on GFE 2.10. I'll look again now to see if I can send a flag value that turns this off.

In any case, when resoIution changes, I think the OMX video pipeline needs to be reset or a port settings changed event needs to be read again to recover when the resolution changes. On Android (which uses OMX), I see a port settings change event printed in the log.

closed 2016-02-28 by irtimmer

irtimmer, 2016-02-28:
Duplicate of #315

PR #322 (2016-02-28): Update common to disable dynamic resolution switching
This should fix #320 and #315