moonlight-stream/moonlight-embedded #321: Failed to pair to server: HTTP response code said error

aikoncwd, 2016-02-28:
I'm using **Moonlight-Embedded 2.1.3**
I'm using **nVidia GeForce Experience 2.10.2.40**
No firewall configured on my PC, no Antivirus is running or blocking any app/process

This is what I get: http://i.imgur.com/Fag4xrp.jpg

I try to pair using `moonlight pair` then I enter the correct PIN into the GFE server (no more popups shows on the GFE server), after a fex seconds moonlight sais: **Failed to pair to server: HTTP response code said error**

Few weeks ago I was able to use moonlight, what happened? Thanks for your time :)

djrockgt, 2016-02-28:
It was a bad graphics driver for me. Accidentally upgrading to the 2.11.x.x beta broke pairing even after downgrading back to 2.10.2.40. Use the steps here to get a clean GFE install and see what happens:

https://github.com/irtimmer/moonlight-embedded/issues/314

Might get you up and running before a dev has a moment to work tickets.

aikoncwd, 2016-02-28:
@djrockgt wicht comment on #314 had the solution? I can't find it :sob:

aikoncwd, 2016-02-28:
It doesn't work :cry: 

I had the beta 2.11 too, then rolledback to 2.10.xx but still with the same error. I followed #314:

1. Download DDU, executing in safe-mode
2. Click on `Clean and restart`
3. Install '361.91-desktop-win10-64bit-international-whql'
4. This driver came with GFE 2.9.xx, tried moonlight and get same error
5. GFE 2.9.xx auto-updated to 2.10.2.40, tried moonlight and get same error

What can I now?

cgutman, 2016-02-29:
This is the pairing bug that is fixed in master. We need a new release to fix this in the Debian and AUR repos.

aikoncwd, 2016-02-29:
@cgutman Hi, when this release will be published? I want to get my moonlight working tomorrow, because FarCry Primal will be relased March 1. Is that possible? Thanks.

closed 2016-02-29 by irtimmer

aikoncwd, 2016-02-29:
Errr... My issue is still on. I can't pair with moonlight.

wozzo, 2016-03-01:
What's the resolution please? @irtimmer

cgutman, 2016-03-01:
@aikoncwd You should direct any further comments to #324. Technically the bug is fixed, but the fix hasn't been released.