moonlight-stream/moonlight-embedded #323: Failed to pair to server: HTTP response code said error

wozzo, 2016-02-29:
When I attempt to pair I get the pin code, switch to GEF server and type it in to the pop up. Click Connect, back to the Pi and it says "Failed to pair to server: HTTP response code said error"

I'm using GEF 2.10.2.40 and have an Nvidia 970M.
For moonlight I have a Pi B+ running Jessie

wozzo, 2016-02-29:
Just to be really confusing if I don't enter the pin and hit decline it now says successfully paired. Either way if I then try moonlight list IP I get a "You must pair with the PC first" message

aikoncwd, 2016-02-29:
I have exactly the same issue here #231, but @irtimmer closed the issue :disappointed:

wozzo, 2016-02-29:
Agreed. Definitely the same issue, and I've never had a beta version of GFE installed and have driver 361.91

irtimmer, 2016-02-29:
Duplicate of #321

closed 2016-02-29 by irtimmer

wozzo, 2016-02-29:
Which also doesn't appear to be resolved? Can you explain the resolution if closing?

wozzo, 2016-03-01:
What's the resolution please? @irtimmer

cgutman, 2016-03-01:
@wozzo #324 describes the task of actually releasing the fixed build to package repositories.