moonlight-stream/moonlight-embedded #324: Packages in AUR and Raspbian/OSMC repositories are broken

cgutman, 2016-03-01:
This is a meta issue that describes a couple major issues with 2.1.3 as released in the official package repositories. I've linked relevant issues under each heading. This should hopefully reduce the duplicate issues being filed.

While these issues have been fixed in master, the fixes haven't been released to the package repositories, which is what this bug describes.

**Unable to pair to any GFE version**
- #321 
- #323 

**Video freezes after a short time (GFE 2.10)**
- #315 
- #320

closed 2016-03-01 by irtimmer

aikoncwd, 2016-03-01:
Thanks! testing right now...

aikoncwd, 2016-03-01:
@irtimmer @cgutman I can't test the new release 2.1.4, after doing
`sudo apt-get update`
`sudo apt-get upgrade`
moonlight is not updated, still stuck in 2.1.3

Anyway, thanks for your work and your time guys! :)

meska, 2016-03-01:
@aikoncwd 
```
wget http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.1.4-1jessie1_armhf.deb
sudo dpkg -i moonlight-embedded_2.1.4-1jessie1_armhf.deb
```

cgutman, 2016-03-01:
I was just looking at this too.

@irtimmer It looks like the Release file in the package repo for Debian Jessie needs to be updated. Wheezy is updated to point to 3.1.4.

aikoncwd, 2016-03-01:
@meska **FUCKING YES!**

https://i.imgur.com/ymV0U8F.png

Time to burn my FarCry Primal over RaspberryPi, thanks all of you guys.

@cgutman gotcha! :smile: