moonlight-stream/moonlight-embedded #326: Cant install Moonlight

BuzzKilll, 2016-03-02:
Hi, when i attempt to install moonlight, i get this error while running the command sudo apt-get install moonlight-embedded on my raspberrypi this error comes up, `sudo apt-get install moonlight-embedded
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  moonlight-embedded
0 upgraded, 1 newly installed, 0 to remove and 75 not upgraded.
Need to get 106 kB of archives.
After this operation, 289 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  moonlight-embedded
Install these packages without verification [y/N]? y
Err http://archive.itimmer.nl/raspbian/moonlight/ wheezy/main moonlight-embedded armhf 2.1.3-1wheezy1
  404  Not Found
Failed to fetch http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.1.3-1wheezy1_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
`

dead, 2016-03-02:
Try update the repositories before.
`sudo apt-get update`
then
`sudo apt-get install moonlight-embedded`

BuzzKilll, 2016-03-02:
I did, still isnt working.

dead, 2016-03-02:
Well, then install the deb from: http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.1.4-1wheezy1_armhf.deb

BuzzKilll, 2016-03-03:
Well, not so surprised that worked Thanks for the great help

closed 2016-03-03 by BuzzKilll