moonlight-stream/moonlight-embedded #327: No Sound

BuzzKilll, 2016-03-05:
When i  stream, i get video but no audio, i uncommented hdmi_drive=2 in /boot/config.txt/ anyway, im using hdmi and wifi, no ethernet sadly? any help please?

Unknownforce351, 2016-03-06:
Try adding dtparam=audio=on to /boot/config.txt

BuzzKilll, 2016-03-07:
Didnt work

KWottrich, 2016-03-07:
Have you tried streaming different games?  I was getting no sound when I tried to stream a game using the Dolphin emulator, but sound worked when I tried a different game.  (The solution for Dolphin audio, if it matters, was to switch to OpenAL for sound.)

dead, 2016-03-07:
You need provide more info...
What command line are you using to run the moonlight?
It is a raspberry pi? 1? 2? What distro?
Are you using an external audio?
What audio backend are you using?

Anyway, take a look a this https://github.com/dead/moonlight-openelec-rpi2/issues/6#issuecomment-192888433 maybe it could help you.

BuzzKilll, 2016-03-07:
Hi, even the default moonlight stream 192.168.1.2 dosent give out any sound, raspberry pi B+, raspian,and using hdmi and tv speakers, im using a custom app in gamestream which is my desktop

Unknownforce351, 2016-03-07:
Let's start from ground zero... Do you get audio from other applications on the Pi?  Not sure what else you might have on your Pi, but does anything else output sound?

Try doing this from the terminal:  speaker-test -c2

Getting sound from this?

BuzzKilll, 2016-03-07:
Yep

Unknownforce351, 2016-03-07:
No errors during compile, correct?  Do you hear the sound from the host computer while streaming?

BuzzKilll, 2016-03-07:
Yeah i do, no errors

Unknownforce351, 2016-03-07:
That's not working as expected.  During stream, audio should be redirected to client, not playing on Host at all.  Try using this option -audio sysdefault

Also, post the output from this command (lowercase L):
`aplay -l`
and from this one as well (uppercase L):
`aplay -L`

BuzzKilll, 2016-03-07:
-adio sysdeafult dosent work : http://pastebin.com/V5JFVg8A

Unknownforce351, 2016-03-07:
how about this output:
`moonlight list`

And what game(s) have you tested with?

BuzzKilll, 2016-03-07:
I cant play any games, my geforce experience just wont update, just gives an error that it cannot connect to nvidea

Unknownforce351, 2016-03-07:
You said you get video but no audio... I'm confused at the problem here...

BuzzKilll, 2016-03-07:
no audio. thats all, i can stream my desktop but no games.

BuzzKilll, 2016-03-07:
i get no audio via steam big picture or via desktop

Unknownforce351, 2016-03-07:
What version of Geforce experience are you running?

BuzzKilll, 2016-03-07:
v2.1.5.0

Unknownforce351, 2016-03-07:
I'm surprised it does anything at all, you need to update this.  http://www.geforce.com/geforce-experience/download

Unknownforce351, 2016-03-07:
And even update the Video drivers probably too... http://www.nvidia.com/Download/index.aspx?lang=en-us

BuzzKilll, 2016-03-07:
I wont be able to reply in about 20 mins or so, going out to a business dinner

BuzzKilll, 2016-03-07:
but ill download them while im gone! :D thanks so far for the help

BuzzKilll, 2016-03-07:
GeForce experience Works!, im updating drivers atm! Thanks ill get back to you soon!

BuzzKilll, 2016-03-07:
IT WORKS, IT WAS MY AUDIO DRIVER!  thanks again! bye bye now!

closed 2016-03-07 by BuzzKilll