moonlight-stream/moonlight-embedded #328: cec support on osmc

sheldon, 2016-03-05:
Hi

I'm having trouble getting CEC to work on OSMC running on a raspberry pi 2.

I tried installing the package version, but CEC didn't seem to work.

Next I tried a compile. Compile worked out the box, but didn't seem to be finding libcec.

I went digging into the cmake setup. On master I applied the following patch to be able to specify where libcec is on my system. My libcec is in ```/usr/osmc/lib``` and ```/usr/osmc/include``` from the rbp2-libcec-osmc and rbp2-libcec-dev-osmc packages.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0beb0b7..a9bc622 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -17,6 +17,7 @@ find_package(ALSA REQUIRED)
 find_package(Opus REQUIRED)
 find_package(Broadcom)
 find_package(Freescale)
+find_package(CEC)

 find_package(PkgConfig REQUIRED)
 pkg_check_modules(EVDEV REQUIRED libevdev)
@@ -27,7 +28,6 @@ pkg_check_modules(AVUTIL libavutil)
 pkg_check_modules(XLIB x11-xcb)
 pkg_check_modules(LIBVA vdpau)
 pkg_check_modules(PULSE libpulse-simple)
-pkg_check_modules(CEC libcec>=3.0.0)

 if(AVCODEC_FOUND AND AVUTIL_FOUND AND SDL_FOUND)
   set(SOFTWARE_FOUND TRUE)
```

During cmake is says it finds libcec ok, but then during compile it fails.

```
In file included from /home/osmc/dev/moonlight-embedded/src/input/cec.c:24:0:
/home/osmc/dev/moonlight-embedded/./third_party/libcec/ceccloader.h:34:18: fatal error: cecc.h: No such file or directory
 #include "cecc.h"
```

I don't know build systems very well, is there any way I can patch the build to find libcec on my sytem correctly?

Here's my issue template answers too:

- [x] It's a bug in Moonlight Embedded and not in NVidia Geforce Experience or Steam as we otherwise can't fix them
- [x] It's not a misconfiguration of your own setup. Like firewall misconfiguration.

I think this might very well be something I'm doing wrong, but I've tried various things now for a few hours and can't seem to fix it.

- [x] Their is no other bug report with the same issue. Check also the closed issues in case your bug is already solved in master.

There's a few reports mentioning cec, but nothing about it not compiling.

Please provide the following info.

**NVidia Geforce Experience version:** _2.10.2.40_
**Moonlight Embedded version:** _2.1.4 & master branch_
**Moonlight Embedded running on:** _Raspberry Pi 2_

**Output of Moonlight Embedded:**

n/a

**What is the expected result?**

Working CEC with the package version, or a successful compile.

**What happens instead of that?**

CEC doesn't seem to work on the package version, and libcec isn't listed as a dependency so I guess it's not in the package version yet?

CEC compiled support working.

Unknownforce351, 2016-03-08:
This difference is bad.  Not sure why it's removed?  You need this line for it to find libcec properly.
`-pkg_check_modules(CEC libcec>=3.0.0)`

The following is no longer needed:
`find_package(CEC)`

If still having problems compiling, then try removing all cec, updating lists and OS, and then re-installing libcec, I actually had to do this recently as well due to my kodi's CEC compile messing with moonlight:
`sudo apt-get remove libcec libcec3
sudo apt-get autoremove
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libcec`

sheldon, 2016-03-13:
@Unknownforce351 thanks for the heads up, I'll give that a try

closed 2016-04-03 by irtimmer