moonlight-stream/moonlight-embedded #330: xbox360 controller problem in game

Kheiryk, 2016-03-06:
Hello,

NVidia Geforce Experience version: 2.1.2.40
Moonlight Embedded version: 2.1.4
Moonlight Embedded running on: RPI2

I've a problem with my controller xbox360 wire, when i check the controls with the command : 
sudo xboxdrv
no problem, the controls appears

I start steam with this command :
**moonlight stream ip -mapping /home/pi/xbox.map -1080 -app "Steam"**

in my rc.local
**xboxdrv --trigger-as-button --wid 0 --led 2 --detach-kernel-driver --deadzone 4000 --dpad-rotation 90 --axismap -DPAD_X=DPAD_X --silent &**

and i've an xbox.map with my controller config (works in 2.1.2)
the game start, but the controller isn't recognized.

if you can help me please
thanks

Kheiryk, 2016-03-06:
I've reboot my pi and my computer, but now

moonlight pair 192...
Moonlight Embedded 2.1.4 (EMBEDDED;PI)
Connect to 192...
Can't connect to server 192...

:'(

shayaantx, 2016-03-07:
For your 2nd problem:

Can your pi ping your host machine? Can your host machine ping your pi?

On your host machine, do you get results for either or the below URL's?
http://127.0.0.1:47989/serverinfo?uniqueid=1234
https://127.0.0.1:47984/serverinfo?uniqueid=1234

------------------------------------------------------------------------------

For your first problem:

Why are you on such an old version for the GFE server? I'm on 2.9.1.22, which has a lot of updates and is fairly stable for me.

Does the controller work directly on your host machine?

Have you tried any other games?

You said "xbox360 wire", you mean wired or wireless? Cause you have -wid, which is for wireless, if you are wired, probably should remove that.

Kheiryk, 2016-03-07:
Hi

> Why are you on such an old version for the GFE server? I'm on 2.9.1.22, which has a lot of updates and is fairly stable for me.

sorry for the error, i've edit my first post, i'm in 2.10, the last version

> Does the controller work directly on your host machine?
yes

> Have you tried any other games?
yes, without steam, same problem

> You said "xbox360 wire", you mean wired or wireless? Cause you have -wid, which is for wireless, if you are wired, probably should remove that.
wired, it's right

thanks

KWottrich, 2016-03-07:
> wired, it's right
thanks

Does that mean the issue was fixed when you removed the --wid from the arguments?  If it was fixed, great!  Please indicate so the issue can be closed.

If not, I suggest trying to run moonlight without the "-mapping /home/pi/xbox.map" argument.  For me, it's enough to use xboxdrv to first map the controller (I also use a wired Xbox 360 controller), then just run moonlight without any special controller mappings.

Kheiryk, 2016-03-07:
Hi

> removed the --wid 

now, i've 
[image]
when i reboot

i must execute **sudo rmmod xpad**

but my problem of pair with the host isn't resolve :/
i can't test in a game

Unknownforce351, 2016-03-07:
You shouldn't need to re-pair the computer/pi just because of a reboot.  You should just be able to stream again.  No need to pair again.

Unknownforce351, 2016-03-07:
Oh, and you would remove "--wid 0" not just the "--wid" part.

Kheiryk, 2016-03-07:
yes, i've remove --wid 0

> You shouldn't need to re-pair the computer/pi just because of a reboot. You should just be able to stream again. No need to pair again.

can't connect

Unknownforce351, 2016-03-07:
In GFE, go to the SHIELD part and click Forget Devices or whatever the button is.  Then reboot the computer and try to pair again.

Kheiryk, 2016-03-07:
doesn't work :/ strange

Unknownforce351, 2016-03-08:
Try installing the latest video drivers and GFE on the host, performing a clean install as well, and updating to latest master git commit, looks like some pairing things were updated a bit.

Kheiryk, 2016-03-13:
Hello
i've reinstall but always the same problem, no paired :s

Kheiryk, 2016-03-13:
Hi, for information, i've install the last beta GFE and it's work.

dead, 2016-03-13:
I think you will need to wait until moonlight 2.1.5 is released.

Kheiryk, 2016-03-14:
ok thanks, i wait :/

dead, 2016-03-14:
You can try compile it yourself, latest branch works with GFE beta. What OS are you using? Raspbian Jessy?

Kheiryk, 2016-03-15:
Hi
Jessy yes
do you have a procedure please ?

Kheiryk, 2016-03-18:
Hello
i've make a new install of moonlight, but it's always the same error. an idea ? please

dead, 2016-03-19:
Take a look at: https://github.com/irtimmer/moonlight-embedded/issues/314#issuecomment-189374108
Download the [UDD](http://www.guru3d.com/files-details/display-driver-uninstaller-download.html)

Kheiryk, 2016-03-20:
Thank you ! I try this :)

Kheiryk, 2016-03-20:
it's done, same problem
i must update with beta GFE for paired and when i make this error 404 ...

I wait the 2.1.5, maybe better in my case ^^
thanks

Kheiryk, 2016-04-06:
**NVidia Geforce Experience version: 2.11.2.55**
**Moonlight Embedded version: 2.2**
**Moonlight Embedded running on: Raspberry Pi 2 / Raspbian Jessy**

Hello, 
With the next version of Moonlight, i've resolved my problem for pair, but i've always my problem with my wired pad.

I start Raspbian and i check with **sudo xboxdrv **
[image]

I must killall xobxdrv and execute **sudo rmmod xpad** for resolve it, after, i can push the buttons on my pad and the commands are displayed to the screen.

But, in game, i can't used my pad, nothing moves
In my rc.local, today
**xboxdrv --trigger-as-button --id 0 --led 2 --detach-kernel-driver --deadzone 4000 --silent &
sleep 1**

for start a game in example
**moonlight stream ip -1080 -app "Steam"**

Thanks

7hunderbug, 2016-04-07:
According to your first post on this issue, your Xbox 360 wireless controller is using the /dev/input/event2 device handle, which means you have more than one input device connected to your Raspberry Pi. This is fine however, if you don't specify the input device for moonlight-embedded, it defaults to /dev/input/event0 (as far as I know). In all your posts here, you haven't yet said you run moonlight-embedded with the -input switch. This is likely the cause of your issue. 

To fix your problem you likely want to run moonlight-embedded like so:

`moonlight-embedded stream -mapping /home/pi/xbox.map -input /dev/input/event2 -1080 -app "Steam" IP.address`

Also, if you don't use the xpad driver at all, you can prevent it from loading into the kernel at boot time by adding
`blacklist xpad` 
to /etc/modprobe.d/xpad-blacklist.conf (create this file because it won't exist)

Another thing; you won't need `--dpad-rotation 90 --axismap -DPAD_X=DPAD_X` arguments for xboxdrv (especially if you make your own mapping file - be sure to make a new moonlight-embedded mapping file if you remove these arguments from xboxdrv though).

Be careful which /dev/input/event* device you use with the -input argument for moonlight-embedded. On my Raspberry Pi with a keyboard and mouse plugged in and with xboxdrv run from rc.local, sometimes the kernel enumerates the xboxdrv input as event0 and sometimes as event2.

Kheiryk, 2016-04-07:
Hello
I've create a file with blacklist xpad in **/etc/modprobe.d/xpad-blacklist.conf **
and i've change my game start bash
**moonlight stream ip -mapping /home/pi/xbox.map -input /dev/input/event1 -1080 -app  "Street Fighter V"**

but in my input, i've
by-id  by-path  **event0  event1**  mice  mouse0
event2 isn't find, i've try with 0 and 1, not works :/

and i've always the problem with the screenshot at the start

Thank you

7hunderbug, 2016-04-07:
xboxdrv needs to be running for it to create an input event handler. In your example, you stopped xboxdrv then `ls` the directory, so of course there is no /dev/input/event2

Either set xboxdrv running by putting it in your /etc/rc.local file or by running it on the command line first before running moonlight. You can keep xboxdrv running by putting `&` at the end of the command line.

Also, this might make a difference: put the GFE host IP address at the end of your moonlight-embedded command line not before the -mapping or -input switches. Like this:

`moonlight stream -mapping /home/pi/xbox.map -input /dev/input/event2 -1080 -app "Street Fighter V" ip`

Kheiryk, 2016-04-09:
Hi

without kill xboxdrv
[image]

but doesn't work
in my rc.local, i've
xboxdrv --trigger-as-button --id 0 --led 2 --detach-kernel-driver --deadzone 4000 --silent &
sleep 1

thanks

Kheiryk, 2016-04-09:
I must update moonlight-common-c for resolve this ?
how i can make this please

Unknownforce351, 2016-04-11:
in rc.local file, add **rmmod xpad** above your **xboxdrv ...**

Unknownforce351, 2016-04-11:
Also, not sure why you're using **--detach-kernel-driver**?

Kheiryk, 2016-04-12:
thanks but doesn't work with and without --detach-kernel-driver and with add "rmmod xpad" before

closed 2017-03-18 by irtimmer