moonlight-stream/moonlight-embedded #332: Can't pair rPI2 with PC: "failed to connect"

Roet-Ivar, 2016-03-08:
Info:
I use the latest version of GeForce Experience 2.10.2 on PC and RecalboxOS 4.0.0-beta1 on the Raspberry pi 2.

What I've tested: 
- I've added my IP in the moonlight config.
- When running: "./Moonlight.sh pair" I get the pairing code sent to my PC (connections seems fine), but after ~10s I get "Failed to connect". 
- There is no "keyfolder" created from any attempts that I can delete either since I never got further than this.

Any ideas?

Roet-Ivar, 2016-03-08:
@irtimmer  Hi sir. Sorry for the duplicate. 
If I get this right:
You have solved this issue in the coming realease of Moonlight-embedded?
If so, then it's up to Recalbox devs to update the bundled version of Moonlight in their next release?

dead, 2016-03-08:
Moonlight 2.1.3 works fine with GFE 2.10...
Try compiling it yourself.

Roet-Ivar, 2016-03-08:
I tested the latest beta driver of GFE and it did pair once. After a reboot the pairing seems lost so I gave it a clean and pair, and again: "Failed to pair to server: Error".

Roet-Ivar, 2016-03-10:
OK guys. It seems like the pairing is OK after using latest GPU drivers and GFE beta if running "clean" first and then "pair".

My next problem is that when starting a stream, only the host seem to render the game while the client (rpi2) doesn't show anything other than the splash screen of Recalbox "v4.0.0-build-10 2016/02/32 20:10". No key input is registered from the client either.

I guess this is another issue that might not be related to this one, so I guess we can close this one. 
I'll have make some research for my next problem.

Thanks guys!

substring, 2016-03-17:
@Roet-Ivar Moonlight 2.1.4 solved all my pairing troubles + my users' problems too. Have you given it a try ? Consider closing this issue if it worked fine

irtimmer, 2016-04-03:
Pairing issues should be resolved with latest release.

closed 2016-04-03 by irtimmer