moonlight-stream/moonlight-embedded #333: Can't pair rPI2 with PC: "failed to connect"

Roet-Ivar, 2016-03-08:
Info:
I use the latest version of GeForce Experience 2.10.2 on PC and RecalboxOS 4.0.0-beta1 on the Raspberry pi 2.

What I've tested: 
- I've added my IP in the moonlight config.
- When running: "./Moonlight.sh pair" I get the pairing code sent to my PC (connections seems fine), but after ~10s I get "Failed to connect". 
- There is no "keyfolder" created from any attempts that I can delete since I never got further than this.

Any ideas?

Roet-Ivar, 2016-03-08:
From Recalbox website:
"GFE 2.10 + Moonlight 2.1.3 seem not to be doing well together yet. Prefer GFE 2.9 if possible"
So I guess I'll have to downgrade my GFE to 2.9 and test again.

I'll come back with results later, please keep this issue open till then.

irtimmer, 2016-03-08:
Duplicate of #332

closed 2016-03-08 by irtimmer

Roet-Ivar, 2016-03-08:
@irtimmer Hi sir. Sorry for the duplicate. If I get this right:
You have solved this issue in the coming realease of Moonlight-embedded?
If so, then it's up to Recalbox devs to update their bundled version of Moonlight?