moonlight-stream/moonlight-embedded #334: RTSP OPTIONS request failed: 11

dhruvparpia, 2016-03-08:
Hi, 

I see to be having issues with the latest version, I've been able to stream using the same settings prior to this release, but now I just get a RTSP handshake error. Anyone else facing the same issue?

**Windows Version:** Windows 10
**NVidia Geforce Experience version:** 2.11.1.2
**Moonlight Embedded version:** 2.1.4
**Moonlight Embedded running on:** Raspberry Pi 2

##### Output of Moonlight Embedded:
    Moonlight Embedded 2.1.4 (EMBEDDED;PI)
    Connect to 192.168.1.5...
    Stream 1920 x 1080, 60 fps, 20000 kbps
    Initializing platform...done
    Resolving host name...done
    Starting RTSP handshake...select() timed out after 10 seconds
    RTSP OPTIONS request failed: 11
    failed: 11

irtimmer, 2016-03-08:
You've installed a beta version of GFE which isn't supported in Moonlight Embedded 2.1.4. But it will be supported in the next release.

closed 2016-03-08 by irtimmer

TheCodeDestroyer, 2016-03-30:
@irtimmer just a question, when is the new release planned?