moonlight-stream/moonlight-embedded #335: Moonlight stream: failed: -1

DatScreamer, 2016-03-10:
Please provide the following info.

**_NVidia Geforce Experience version:210.40
**Moonlight Embedded version:2.1.2 (Obtained through the raspberry pi guy's [install.sh](https://github.com/the-raspberry-pi-guy/game_stream/blob/master/install.sh) file
**Moonlight Embedded running on:Raspberry Pi & Raspbian (From NOOBS Latest)

**Output of Moonlight Embedded:**
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 0
failed: -1

**What is the expected result?**
Expected result of moonlight stream -720 -30fps -app tankix is Start Stream
**What happens instead of that?**
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 0
failed: -1

On another note when trying to install Moonlight without TRPG's install.sh file I get:
unzip moonlight-embedded-master.zip
cd moonlight-embedded-master
mkdir build
cd build
cmake ../

CMake output:
-- The C compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.28") 
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so  
-- Found Broadcom: /opt/vc/include  
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'libevdev'
--   found libevdev, version 1.4.3
-- checking for module 'libudev'
--   found libudev, version 215
-- checking for module 'sdl2>=2.0.4'
--   package 'sdl2>=2.0.4' not found
-- checking for module 'libavcodec'
--   package 'libavcodec' not found
-- checking for module 'libavutil'
--   package 'libavutil' not found
-- checking for module 'x11-xcb'
--   package 'x11-xcb' not found
-- checking for module 'vdpau'
--   package 'vdpau' not found
-- checking for module 'libpulse-simple'
--   package 'libpulse-simple' not found
-- checking for module 'libcec>=3.0.0'
--   package 'libcec>=3.0.0' not found
-- checking for one of the modules 'uuid'
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0") 
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1k") 
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.1.0") 
-- checking for module 'avahi-client'
--   found avahi-client, version 0.6.31
-- checking for module 'libenet'
--   package 'libenet' not found
CMake Error at /usr/local/share/cmake-3.3/Modules/FindPkgConfig.cmake:340 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.3/Modules/FindPkgConfig.cmake:502 (_pkg_check_modules_internal)
  libgamestream/CMakeLists.txt:8 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeError.log".

CMakeError.log:
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_12bc8/fast"
/usr/bin/make -f CMakeFiles/cmTC_12bc8.dir/build.make CMakeFiles/cmTC_12bc8.dir/build
make[1]: Entering directory '/home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_12bc8.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_12bc8.dir/CheckSymbolExists.c.o   -c /home/pi/Downloads/moonlight-embedded-master/bui$
Linking C executable cmTC_12bc8
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_12bc8.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_12bc8.dir/CheckSymbolExists.c.o  -o cmTC_12bc8 -rdynamic
CMakeFiles/cmTC_12bc8.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x38): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_12bc8.dir/build.make:97: recipe for target 'cmTC_12bc8' failed
make[1]: Leaving directory '/home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_12bc8/fast' failed
make[1]: *** [cmTC_12bc8] Error 1
make: *** [cmTC_12bc8/fast] Error 2

File /home/pi/Downloads/moonlight-embedded-master/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */

cgutman, 2016-03-10:
The version of Moonlight Embedded in that install script needs to be updated. Ideally, it should just install from the Raspbian package repos now that they exist.

DatScreamer, 2016-03-10:
So what do I replace that line wget ..... with?

I have downloaded the files from github/.../moonlight.../releases and now am getting:
[ 47%] Building C object libgamestream/CMakeFiles/gamestream.dir/client.c.o
/home/pi/Downloads/moonlight-embedded-2.1.3/libgamestream/client.c:32:23: fatal error: uuid/uuid.h: No such file or directory
 #include <uuid/uuid.h>
                       ^
compilation terminated.
libgamestream/CMakeFiles/gamestream.dir/build.make:134: recipe for target 'libgamestream/CMakeFiles/gamestream.dir/client.c.o' failed
make[2]: *** [libgamestream/CMakeFiles/gamestream.dir/client.c.o] Error 1
CMakeFiles/Makefile2:163: recipe for target 'libgamestream/CMakeFiles/gamestream.dir/all' failed
make[1]: *** [libgamestream/CMakeFiles/gamestream.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I have tried this with the latest 2 releases (2.1.4,2.1.3)

DatScreamer, 2016-03-10:
OK, I did what you have said. I figured it out with your help. It is working with that. I am new to GitHub so I did not know that page existed (Noob not knowin how to navigate the website).
Thanks! It is working.

closed 2016-03-10 by irtimmer