moonlight-stream/moonlight-embedded #338: RTSP handshake error

Simber1, 2016-03-15:
**NVidia Geforce Experience version:2.11.1.2 (Beta)**
**Moonlight Embedded version:2.1.4**
**Moonlight Embedded running on:Pi 3**

When trying to connect it keeps giving using moonlight stream -1080 -60fps 192.168.1.214

Starting RTSP handshake...select() timed out after 10 seconds
RTSP OPTIONS request failed: 11
failed:11

Steam still starts on my pc fine and everything acts like it's still streaming

cgutman, 2016-03-15:
Version 2.1.4 is not compatible with GeForce Experience Beta 2.11. You'll need to downgrade to the 2.10 production version or build the latest Moonlight Embedded code which does have support for 2.11.

irtimmer, 2016-03-15:
Duplicate of #334

closed 2016-03-15 by irtimmer