moonlight-stream/moonlight-embedded #340: RTSP OPTIONS request failed: 11

MartinSpacil, 2016-03-16:
I use the latest GeForce Experience "2.10.2.40" on PC (Windows 10, x64, GTX750 Ti). Moonlight (Embedded 2.1.4) is installed on Raspberry Pi 3 (
Raspbian Jessie). Both devices are connected via an ethernet cable.

I can pair devices without any problem. But I can stream content only in 1280x720 resolution and with 30fps.

When I try to stream 1920x1080 with 30/60 fps or 1280x720 with 60fps, I get an error: 
RTSP OPTIONS request failed: 11
failed: 11

Has anybody idea, what's wrong?

Thanks, M.S.

P.S. Whole console output: 
[1drv.ms/1Uz5COk](url)

7hunderbug, 2016-03-16:
See https://github.com/irtimmer/moonlight-embedded/issues/338

7hunderbug, 2016-03-16:
Ah, you're not running GFE beta though perhaps building Moonlight-embedded from source will fix the issue?

Simber1, 2016-03-16:
or See https://github.com/irtimmer/moonlight-embedded/issues/336

MartinSpacil, 2016-03-16:
I will try build Moonlight from source code tonight. Then I send a message.

MartinSpacil, 2016-03-16:
I attempted to build Moonlight from the source code, but without succes.

I get some erros:

1) Package rbp-userland-osmc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or is only available from another source

E: Package 'rbp-userland-osmc' has no installation candidate

2) CMake 3.1 or higher is required.  You are running version 3.0.2

I'm afraid, that I dont have enough time to complete this process... 
I'll have to wait for official release.

M.S.

Simber1, 2016-03-16:
You have to jump through some hoops to get the needed version of cmake. I couldn't get it to compile either.

dead, 2016-03-16:
Apply this [patch ](https://github.com/irtimmer/moonlight-embedded/blob/raspbian/jessie/debian/patches/install.patch) to compile using CMake 3.0

Septiron, 2016-03-17:
Hi, 
i have the same issue and try to fix it for my self in the last three days. 
My settings are:
Windows PC: Windows 8.1 Pro and now Windows 10 Pro x64, Geforce GTX 760 (Driver version: 364.51 - latest), Geforce Experience 2.10.2.40
Raspberry: Debian Jessie Raspbian 2016.02, Moonlight 2.1.4

What i have done so far:
- Set up my Raspberry with Raspbian and Moonlight and try to stream my Windows 8.1 Pro PC:
RTSP OPTIONS request failed: 11
- Reinstall all nVidia Software on my Windows 8.1 Pro PC and tried again: same RTSP error
- Deactivate Windows Firewall and try again (new error code): RTSP OPTIONS request failed: 111
- Uninstall AVG Antivirus (tests with and without Firewall): RTSP OPTIONS request failed: 11 and RTSP OPTIONS request failed: 111
- Reinstall my PC with Windows 10 Pro (fresh installation), installed Avast Antivir, Steam with some games and test it again: The same Errorcode (11) occurs as on the Windows 8.1 installation. After that i reinstalled my Raspbian and test it again. 
- I tried the same steps as on my Windows 8.1 PC with the "new" Windows 10 installation of it and get the same messages. 

Today i found this issue here and i must say, that i don't get it, what i have to do, when you say i have to apply and compile patches. 
Are these patches in the next official release of moonlight? 

Alex

dead, 2016-03-19:
The `RTSP OPTIONS request failed: 11`error occurs when you are using a beta version of GFE, but that strange since you are not using the beta.
Try uninstalling the driver using [DDU](http://www.guru3d.com/files-details/display-driver-uninstaller-download.html) in safe mode. And install the latest "normal" driver, choose "advanced" and check "clean install".

MartinSpacil, 2016-03-22:
I uninstalled GeForce drivers (and GeForce Experience) with DDU. Then I installed latest GeForce drivers. But behaviour is the same:

RTSP OPTIONS request failed: 11
failed: 11

Septiron, 2016-03-22:
I don't know if i tested it wrong, but i found out, that i perhaps makes some mistakes, when i was trying to stream. 

I always test it this way:
`moonlight stream pc-name` --> RTSP error 11
`moonlight stream -1080 pc-name` --> RTSP error 11

But with this it worked for me:
`moonlight stream -60fps -720 -app "Mass Effect(TM) 3"`

So: After the PC is paired with the device i guess you don't have to use the pc name / ip adress for the "stream" option. You also should try it to call with the parameters -60fps -720 and -app "". It won't worked for me, without the -app parameter.

I hope it would work for you.

And many thanks for the development! I just finished the Mass Effect Triloy with moonlight on my 55 inch TV. Really cool - the program and the game :-) *thumbs up*

Greetings Alex

aikoncwd, 2016-03-30:
Just uninstall GFE 2.11 and reinstall GFE 2.10, reboot and error "11" will disappear.

Septiron, 2016-03-30:
There is no GFE beta installed. The RTSP error "11" even occurs on GFE 2.10 randomly. When it apears i restart the pc. It´s ok for me, but i don't know if the error still occurs by Martin.

MartinSpacil, 2016-03-30:
I had some time for experiments today.
So I installed latest GF drivers (364.72) and latest GeForce Experience (2.11.2.55). I can't stream with these versions nothing. No matter if i set resolution 720/1080 or 30/60 fps. I get an error: RTSP OPTIONS request failed: 11, all the time.
But then I returned to the "old" version of GF drivers (358.87) na GF Experience (2.5.15.54). It's the oldest version of drivers bundle on NVIDIA web. And with these versions I can stream with every resolution and framerate!
It seems that I have found the solution for me.
Thanks for your ideas and support.

Werther86, 2016-03-31:
Hi everyone,
This problem also applies to me with latest GF drivers (364.72) and latest GeForce Experience (2.11.2.55). 

This morning I was streaming my games properly (not absolutely sure but I assume it was GeForce Experience (2.11.2 or 2.11.2.46) and just a few hours after I was not able to stream any more.
It is much probably because moonlight source code should be modified according to updated GeForce Exp software and I think this may be prevented by disabling auto update ability of GeForce Experience in the future.

If somebody can share:
- how to disable auto update ability of Geforce Experience client 
- how to roll back to previous version of Geforce Experience client
this problem may be temporarily fixed for most of us.

Searching answers to my above questions for a few hours but couldn't find a quick turnaround yet.

hodginsa, 2016-04-01:
Its not the Driver. I was getting the same Error 11 so I downgraded GFE to 2.10 using this link:
http://nvidia-geforce-experience.soft32.com/free-download/ 
and it worked after I uninstalled 2.11 and installed 2.10.

Werther86, 2016-04-01:
> it worked after I uninstalled 2.11 and installed 2.10.

Do you know how to disable auto update of GFE?

Because even if I install 2.10, it updates to the latest version (2.11.2.55) automatically. Couldn't find the way disabling this.

cgutman, 2016-04-01:
The changes are ready to go in #351 once @irtimmer has time to merge and release

hodginsa, 2016-04-01:
> Do you know how to disable auto update of GFE?

> Because even if I install 2.10, it updates to the latest version (2.11.2.55) automatically. Couldn't find the way disabling this.

Yeah it literally just did this to me. When I first installed it, it opened and ran 2.10, but then when I opened GFE this morning it installed the update automatically. .

Werther86, 2016-04-02:
> it literally just did this to me.

Unfortunately it happens.
@cgutman thanks for the change but it is not clear to me. Is moonlight ready to update or not? When I try to install moonlight-embedded on Raspbien it says there is no update on that package. How will we apply this fix?

Septiron, 2016-04-02:
Since auto update of GFE 2.11 the RTSP error 11 occurs again. =(

filledk, 2016-04-02:
any fixes to this? im having the same problem :-(

cgutman, 2016-04-02:
@irtimmer waiting for any more fixes to land before releasing 2.1.5? Both production and beta channels of GFE are broken with 2.1.4.

irtimmer, 2016-04-02:
My ARM buildsystem is currently broken so I could only do a source release which isn't very usefull for most users. Beside I still have a lot of code which I want to include in the next release.

Werther86, 2016-04-03:
For those having the same problem...

It seems that I managed to find a way stopping GFE Auto update thanks to user "KnifMelti" in the following: https://forums.geforce.com/default/topic/857331/stop-geforce-experience-from-updating/?offset=5

Long story short:
1) Rollback to the latest working GFE Client 
2) Stop GFE Updating automatically by adding its services URL into your hosts-file as:
127.0.0.1 services.gfe.nvidia.com

so GFE on your host PC will continuously search an update locally.

Didn't try it yet as don't have a time today. But for others seeking a sunday evening full of games then give it a shot.

If we don't stop GFE automatic update, even if irtimmer fixes the issue, we may encounter it again tomorrow.

So hope it works.

PS: I appreciate if somebody can share the latest working GFE setup (before 2.11.2.55) link. Not sure which it was; I have v2.10.2.40 setup latest.

irtimmer, 2016-04-03:
Fixes for GFE 2.11 have landed in Moonlight Embedded

closed 2016-04-03 by irtimmer