moonlight-stream/moonlight-embedded #341: Can't Pair with my Pc Gaming

ElScratou, 2016-03-17:
Hi everyone,

I follow the different steps to this tutoriel [http://www.howtogeek.com/220969/turn-a-raspberry-pi-into-a-steam-machine-with-moonlight/].

And i can't pairing my pc gaming with my raspberry pi 3.

See my configuration : 
NVidia Geforce Experience version: 2.10.2.40
Moonlight Embedded version: 2.1.4

pi@...... ; ~ $ moonlight pair X.X.X.X (<= change by my local adress)
Moonlight Embedded 2.1.4 (EMBEDDED;PI)
Connect to X.X.X.X.....
(wait some time) and finally get
Can't connect to server X.X.X.X (and any popup show up in my pc gaming desktop)

Output of Moonlight Embedded: Can't connect to server X.X.X.X, i'm sure of my IP adresse because i ping my device and i check it with ifconfig/ipconfig

What happens instead of that? Moonlight can't connect and i can't do the final step : "Enter the PIN to complete the pairing process and authorize the Moonlight/Pi unit to access your game stream"

substring, 2016-03-17:
just to be sure, you can do a `avahi_browse -tap _nvstream._tcp` to make sure your GFE host is visible and get its ip

ElScratou, 2016-03-17:
Result of the avahi_browse command : 

pi@raspberry : ~ $ avahi-browse -t _nvstream._tcp
+  wlan0 IPv4 SCRATOUPC                                     _nvstream._tcp       local

ElScratou, 2016-03-22:
Anyone to help me ?!

spacewave, 2016-03-29:
i had issues when the nvidia control panel wasn't in my system tray.  
I hate that stupid icon.  Took me forever to figure that out.

irtimmer, 2016-04-03:
Duplicate of #309

closed 2016-04-03 by irtimmer

ElScratou, 2016-04-29:
No it's no duplicate i can ping my computer and my pi each other.

But when i do the pairing command it takes ages an i haven't got any result or digit to tape on my computer.